chili-chips-ba/openCologne #4: PCB Feature Request: Provision hooks for per-rail current measurements

chili-chips-ba, 2024-06-07:
In order to support the upcoming _Power Analysis_ feature in CologneChip design flow, we need to be able to measure supply currents in a precise, straightforward and accessible way. In particular:
- Core supply rail
- All individual I/O Banks
- SerDes supply rail
- PLL supply

The objective is to correlate actual power numbers with tool-predicted metrics, thus helping CologneChip engineers tune and validate tables and algorithms built into their new tool.

@DadoCCAG can provide additional detail and requirements...

chili-chips-ba, 2024-11-03:
@pu-cc @DadoCCAG how realistic are the CologneChip plans to offer a power estimation tool for GateMate? If that's a far fetched objective, we could close this issue, to then reconsider it for a follow-up project, when CologneChip is ready.

chili-chips-ba, 2024-11-06:
Closing due to lack of interest from Cologne Chip side. We can reopen it later on, when and if the situation changes.

closed 2024-11-06 by chili-chips-ba

pu-cc, 2024-12-05:
We have already implemented power estimation - it is of great interest to us - and are in the test phase. The implementation of this was not something that could be worked out here.