chili-chips-ba/openCologne #10: GateMate Constraint File Pin Declaration Format

TarikHamedovic, 2024-06-08:
The current constraint format for GateMate chips requires the declaration of pin directions both in the constraint file and the HDL file such as `Pin_in`, `Pin_out`, `Pin_inout`. This approach is unconventional and differs from other vendors, creating challenges in implementing Amaranth HLS for GateMate chips. The primary issue is the inability to create a general constraint file due to the necessity of specifying pin directions in the constraint file.
### Proposed Solution

To resolve this issue, it is suggested to use a general pin constraint format that only requires the pin identifier, without the need to specify its direction in the constraint file. This would align with common practices and facilitate easier implementation and integration with Amaranth HLS. 

By adopting this improvement, the workflow for developing with GateMate chips would become more streamlined and user-friendly, aligning better with industry standards.

chili-chips-ba, 2024-06-08:
@pu-cc for awareness

chili-chips-ba, 2024-06-09:
... given [Yosys SV](https://github.com/YosysHQ/yosys/issues/4436#issuecomment-2156815456) problems, one option to consider is to move all new design creation to Amaranth. However, while that's a no brainer for other vendor's devices, these CologneChip constraints peculiarities stand in the way of adopting such development flow to GateMate.

chili-chips-ba, 2024-06-17:
Team, any movement on this development track?!

whitequark, 2024-06-17:
I am not a part of your team nor am I responsible for addressing this issue.

chili-chips-ba, 2024-07-12:
@DadoCCAG , given that it seems very difficult for Amaranth to adapt to this GateMate uniqueness of the constraints' format, what's your assessment from CologneChip side on the chance for providing an option that would align it with other FPGA vendors?

The objective is to enable GateMate in Amaranth flow, making the first critical step towards HLS, and opening a whole new set of possible applications.

pu-cc, 2024-07-17:
With the upcoming P&R update, CCF will now offer the option of specifying the keyword “NET” instead of the pin direction. P&R then retrieves the direction information from the buffer primitives in the netlist. This is compatible with the UCF file format.

As soon as the update is online, I will report it here to close #10.

chili-chips-ba, 2024-07-17:
... that's a wise approach, one that both retains compatibility with CC legacy, and gains compatibility with everything else 💯

pu-cc, 2024-07-19:
We updated the P&R binary yesterday. CCF now supports the “NET” keyword as described [here](https://colognechip.com/downloads/ccgm1a1-evb-master.ccf). You can download the latest build it here:

https://colognechip.com/downloads/cc-toolchain-linux.tar.gz
https://colognechip.com/downloads/cc-toolchain-win.zip

Thanks for your help.

closed 2024-07-19 by pu-cc