chili-chips-ba/openCologne #11: UART Receiver and Transmitter Not Functioning on Olimex GateMate Board

TarikHamedovic, 2024-06-08:
## Description

While testing the [UART receiver and transmitter](https://github.com/chili-chips-ba/openCologne/tree/main/2.Simple--5--UART-Test) on the Olimex GateMate board, we encountered an issue where input typed in the terminal was neither received nor transmitted.
## Steps Taken

1. *Firmware Check*: Verified the RP2040 firmware to ensure UART communication was enabled. The firmware confirmed that UART communication was enabled.
2. *Documentation Review*: Consulted the[ documentation](https://github.com/chili-chips-ba/openCologne/blob/main/0.doc/Olimex/GateMateA1-EVB---Schematic.RevA.pdf) for the Olimex board to verify if the Receive and Transmit Debug pins were declared and connected to the RP2040. The documentation confirmed that the pins were correctly declared and connected.

## Expected Behavior

Input typed in the serial communication terminal should be received and transmitted correctly.

## Actual Behavior

No input typed in the terminal is being received or transmitted.

## Additional Information

- Baud Rate: 115200
- Port: /dev/ttyACM0

TarikHamedovic, 2024-06-08:
The problem was that the documentation for the GateMate Olimex board was wrong. 

The pins were declared like:

```
Pin_out "o_Tx_Serial" Loc = "IO_SA_A6";
Pin_in "i_Rx_Serial" Loc = "IO_SA_B6" | PULLUP=true;
````
Which when looking at the [documentation](https://github.com/chili-chips-ba/openCologne/blob/main/0.doc/Olimex/GateMateA1-EVB---Schematic.RevA.pdf) is correct.

But in actuality they pins are reversed and the correct way to declare them is:
```
Pin_out "o_Tx_Serial" Loc = "IO_SA_B6";
Pin_in "i_Rx_Serial" Loc = "IO_SA_A6" | PULLUP=true;

```

chili-chips-ba, 2024-06-08:
@TsvetanUsunov, take note

closed 2024-06-09 by chili-chips-ba