chili-chips-ba/openCologne #5: PCB Feature Request: Provide good dual analog OpAmp

chili-chips-ba, 2024-06-08:
In order to support our 99% digital, DSP-based _**Sigma-Delta ADC** and **DAC**_ IP cores, we need two board-level analog OpAmps.

- 1st is to be connected as a **comparator**, and used for the SD ADC front-end
- 2nd is to be connected as an **integrator**, and used for the SD DAC final stage

The objective is to offer an on-demand, easy-to-use ADC/DAC solution for CologneChip FPGAs, implemented in simple, affordable, minimal set of external analog components, yet boasting unusually good specs. 

While most of the secret sauce is in the DSP algorithms and tradeoffs between sampling rate, throughput and precision, i.e. custom-tailoring the filtering and averaging functions to the specifics of the use case, this very thin external analog layer must be on par with our digital cores, and contribute to the quality of the total solution.

@DadoCCAG shall provide additional guidance and requirements wrt the selection of this important external component...

chili-chips-ba, 2024-11-03:
@TarikHamedovic could you please add your input here regarding external electronics you've built for Sigma-Delta ADC, as well as radio receiver.

TarikHamedovic, 2024-11-06:
You can view the explanations, schematics and pictures here on the [Wiki](https://github.com/TarikHamedovic/SDR-HLS/wiki/5.Sigma%E2%80%90Delta-Modulator-Implementation) page.

For the radio reciever, this is the [github repo](https://github.com/emard/flearadio/blob/master/doc/FleaFPGA%20Digital%20Radio%20Example%20Schematic.pdf)

This is the schematic:

[image]

This is the PCB layout of the simple board I made:

[image]

And the 3D view of it:

[image]

chili-chips-ba, 2024-12-04:
[MAX40089](https://www.analog.com/media/en/technical-documentation/data-sheets/MAX40077-MAX40089.pdf) is our preference. It comes with full support collateral, from Gerber files to Spice models. While the corresponding [EvalKit](https://www.analog.com/en/resources/evaluation-hardware-and-software/evaluation-boards-kits/max40089evkit.html#eb-overview) is using BNC connectors, we'd rather have the SMAs. 

Full ADC and DAC is also a good idea, ideally of Sigma-Delta type, with 2MSPS+ sampling rate at a least 12-bit wide.

chili-chips-ba, 2024-12-07:
@pu-cc how about posting here links to the Sigma-Delta ADC and DAC designs that you already have for GateMate?!

goran-mahovlic, 2025-05-10:
Board is here, but I did not have time for tests

https://github.com/intergalaktik/Extension_Boards_for_Olimex_GateMate/tree/main/adcIO

closed 2025-05-10 by goran-mahovlic