chili-chips-ba/openCologne #6: PCB Feature Request: Provide sufficient number of LEDs and Push-Buttons, via 1-Wire GPIO expander

chili-chips-ba, 2024-06-08:
While working with Olimex GateMateA1-EVB, we realized that there was only one push-button at user disposal, whereas our TetriSaraj game needed [five](https://github.com/chili-chips-ba/openXC7-TetriSaraj?tab=readme-ov-file#-game-controls-). Similarly the number of user LEDs is inadequate for debugging (@TsvetanUsunov, take note).

We believe that many other examples would also benefit from a richer complement of LEDs and buttons, and recommend providing a PMOD with an 8-bit bank of each. 

Given that this problem is due to GateMate somewhat lower I/O count compared to FPGAs in the same category (@DadoCCAG, take note), these 16 additional lines should come from a **GPIO expander** based on **1-Wire protocol**, and would therefore consume only one FPGA I/O pin for the gain of 16 (or more) slow user I/Os.

To enable this new PMOD hardware, we shall within this project **develop RTL for 1-Wire Controller**.

chili-chips-ba, 2024-11-03:
@goran-mahovlic, the 1-Wire Controller Example is now available:
- https://github.com/chili-chips-ba/openCologne/tree/main/2.Simple--4--1Wire-Controller

goran-mahovlic, 2024-11-04:
Do you maybe have some chip on your mind that we could use? I have only used I2C ones with IRQ capabilities.

goran-mahovlic, 2024-11-04:
DS2408 is really overpriced - on Mouser it is more then 10€

>> We are certain that the Far-Eastern silicon suppliers have their own equivalents as a lower price points. If that option is open, we can also list some from their portfolio.

Sure! I am just preparing orders from LCSC so if something is available there it would be perfect!

chili-chips-ba, 2024-11-05:
As 1-Wire GPIO expanders seem rather scarce, we could amend the plan and build an I/O expander that's truly different from the rest of the crowd. Here are some options:

**---1--- Differentiate by capacity**, such as by offering a large number of I/O
 - [PCA9505](https://www.nxp.com/docs/en/data-sheet/PCA9505_9506.pdf) - 40 GPIO 400kHz I2C, with integrated 100K pullups

**---2--- Differentiate by extra functionality**, such as by offering GPIO with built-in FIFOs, UARTs, PWM, Keypad, PowerWakeup, etc.
 - [SCH3221](https://www.lcsc.com/datasheet/lcsc_datasheet_2203222030_Microchip-Tech-SCH3221-7U_C633298.pdf) - 33 GPIO, Quad 460kbps UARTs, Dual 4Mbps IrDA, Watchdog, Wakeup, PWM
 - [LM8330](https://www.lcsc.com/datasheet/lcsc_datasheet_2303010830_Texas-Instruments-LM8330TME-NOPB_C2862188.pdf) - 20 GPIO with Keypad/PWM controller and power Wakeup
 - [STMPE1601](https://www.lcsc.com/datasheet/lcsc_datasheet_2108071530_STMicroelectronics-STMPE1601TBR_C2669702.pdf) - 16 GPIO with Keypad/PWM controller and power Wakeup
 - [ADP5586](https://www.lcsc.com/datasheet/lcsc_datasheet_2205191300_Analog-Devices-ADP5586ACBZ-01-R7_C659304.pdf) - 10 GPIO with event FIFOs and PWM

 **---3--- Differentiate by speed**, such as by offering 26MHz SPI instead of 400kHz I2C, possibly also with integrated Level Shifters
 - [MAX7301](https://www.lcsc.com/datasheet/lcsc_datasheet_2310091027_Analog-Devices-Inc--Maxim-Integrated-MAX7301AAI-T_C3170288.pdf) - 28 GPIO via 26MHz SPI
  - [MAX7317](https://www.lcsc.com/datasheet/lcsc_datasheet_2304140030_Analog-Devices-Inc--Maxim-Integrated-MAX7317ATE-T_C2652306.pdf) - 10 GPIO via 26MHz SPI, with **oversized 7V/24mA open-drain** outputs
  
  - [XRA1405](https://www.lcsc.com/datasheet/lcsc_datasheet_2108071530_MaxLinear-XRA1405IL24TR-F_C2669710.pdf) - 16 GPIO via 26MHz SPI with Level Shifters
 - [XRA1404](https://www.lcsc.com/datasheet/lcsc_datasheet_2309041631_MaxLinear-XRA1404IL16TR-F_C5838038.pdf) - 8 GPIO via 26MHz SPI with LS
 - [XRA1403](https://www.lcsc.com/datasheet/lcsc_datasheet_2106071032_MaxLinear-XRA1403IG24TR-F_C2669746.pdf) - 16 GPIO via 26MHz SPI, without LS
 
 
 ----
 The **SCH3221** option is very interesting and most challenging.
[image]

goran-mahovlic, 2024-11-05:
Would it be ok if we start with something simple like [PCA9535](https://www.nxp.com/docs/en/data-sheet/PCA9535A.pdf) 

I have that one at home as we have used it in MegaPhone project...

We could use it with few LEDs and Keypad Matrice as a PoC - then if needed per project we could add any of the complex ones?

[PCA9505](https://www.nxp.com/docs/en/data-sheet/PCA9505_9506.pdf) is also fine but I will need to order it, and we need to check what would it be connected to...

I also do not know if it is better for you that we use PMOD form factor for this one or we could use one of Olimex top 3 connectors?

chili-chips-ba, 2024-11-05:
PCA9535 is very popular and also available on [LCSC](https://www.lcsc.com/product-detail/I-O-Expanders_NXP-Semicon-PCA9535CPW-118_C2652292.html). It does the job, but it does not add the uniqueness. 

For the latter, we are proposing to add a [ULN2003](https://www.lcsc.com/datasheet/lcsc_datasheet_2304140030_Texas-Instruments-ULN2003AIDR_C94832.pdf) into the mix. That would also justify the use of top Olimex connector, which has more pins than a PMOD -- We would now not be blocking them without adding value.

chili-chips-ba, 2024-11-07:
We are basically looking for something like [Anvyl](https://digilent.com/reference/learn/programmable-logic/tutorials/anvyl-demo-project-suite/start), but stripped down to the essentials and I/O only, and in the plug-in form, with FPGA on the motherboard.

[image]

chili-chips-ba, 2024-11-15:
@pu-cc any requests or suggestions from your side?

goran-mahovlic, 2025-05-10:
Just for testing purposes much simpler board is done here - we can close this issue and open at some point when we get to level 2 of this project.

https://github.com/intergalaktik/Extension_Boards_for_Olimex_GateMate/tree/main/eduIO

closed 2025-05-10 by goran-mahovlic