chili-chips-ba/openCologne #12: PCB Feature Request: Plug-and-Play compatibility of ULX5M FPGA pinout to Olimex

chili-chips-ba, 2024-06-09:
In order to facilitate porting of apps and example across these two GateMate hardware platforms, ULX5M should strive to match Olimex pinout at FPGA level. With that, the GateMate gateware becomes platform-agnostics, which helps portability and ease of adoption. Even minor pinout discrepancies can cause major trouble, see https://github.com/chili-chips-ba/openCologne/issues/11#issuecomment-2156149819.

The ULX5M connector pinout naturally needs to conform to the CM4 spec. The gist of the **challenge is therefore to map CM4-Connector to Olimex-FPGA pinout on the ULX5M SOM**.

chili-chips-ba, 2024-11-03:
For background, here is the ULX5M concept drawing
[image]

chili-chips-ba, 2024-12-07:
@pu-cc's feedback from today's meeting:
- Ethernet is a the first priority peripheral that CologneChip needs (we plan on using `Realtek RTL8211E`)
- HyperRAM is easy from the PCB viewpoint, but expensive to buy (cca 4 EUR) and difficult to work with. It also has too high latency for many use-cases
- DDR2/3 is a more preferable option
-----------
@goran-mahovlic's input:
- DDR2/3 in unrealistic for GateMate due to higher pin requirements. SDRAM is probably the most we can do. 32MBytes in the 32-bit form-factor is possible (Gowin Aurora devices have 8MByte and 16MByte 32-bit SDRAM options in the SIP physical format)
- **we need to see what to do for GateMate SerDes on the CM4 => This is a call for input!**

goran-mahovlic, 2025-05-10:
I have decided that we will go with SDRAM and Microchip Ethernet chip as Realtek RTL8211E is almost impossible to find.

Board repository is here - https://github.com/intergalaktik/ulx5m-gs

closed 2025-05-10 by goran-mahovlic