chili-chips-ba/openCologne #14: CC_PLL Verilog simulation model bug

tarik-ibrahimovic, 2024-06-10:
The provided simulation Verilog model of CC_PLL isn't a correct representation of the true CC_PLL's behavior. Whatever parameters put in when instantiating CC_PLL get lost and the output clock frequency is just half the input.

This can be especially confusing when running simulation of the examples built in with the tools.

chili-chips-ba, 2024-06-10:
[image]

chili-chips-ba, 2024-07-12:
Hi @pu-cc , given recent finding related to [Issue#17](https://github.com/chili-chips-ba/openCologne/issues/17): `... there is indeed a misconfiguration of a CPE that leads to incorrect inversion...`, could that also be the root cause of this PLL simulation problem?

pu-cc, 2024-07-14:
No, the CPE configuration and the simulation models are completely unrelated to each other.

The PLL model is used exclusively for the simulation and has so far been kept so simple on purpose; of course with the disadvantage that no _real_ frequencies are generated. We have now revised the model for the upcoming update. I will let you know here so that we can close #14. Incidentally, this is not a bug, but a functional enhancement.

pu-cc, 2024-07-19:
We have now improved the simulation model of `CC_PLL` in `bin/p_r/cpelib.v`.

closed 2024-07-19 by pu-cc