chili-chips-ba/openCologne #15: Simple IO expander for Onlimex Misc connector

goran-mahovlic, 2024-06-10:
I would recommend simple IO board that could be used for debugging.

Something like this with 8 LEDs, 4 BTNs and simple audio out.

[image]

But it would be shame to not use SERDES pins on same connector - so let me know if you have some ideas what could I put there.

On ULX4M I have just simple 2.54 header and with that I was able to test one bit BT

https://github.com/newhouseb/onebitbt

chili-chips-ba, 2024-06-11:
## Buttons, LEDs, and other GPIO
Directly-connected Buttons and LEDs are indeed the easiest to work with. However, as GateMate is a bit short on the I/O resources, consider reducing the direct ones to 4 Buttons + 4 LEDs.

The remaining 4+4 could then be connected indirectly, via **1-Wire GPIO** Expander device. This would also open the path for developing 1-Wire Controller for GateMate, which is the serial interface that seems under-represented in the opensource hardware community

## SerDes
As we plan to develop SerDes-based core for connecting two GateMate board (in the likes of Xilinx **Chip2Chip** IP, only super "**Lite**"), considering its 5Gbps speed, we wonder whether ordinary 100mil headers would cut it for multiple hops and decent distances. We are therefore proposing to consider zero-cost **PCIE 1-lane edge connector**
> [image]

We could in such case use standard, off-the-shelf PCIE extender cable for connecting two boards. Please, sync up with @AnttiLukats on this topic, esp. if there is also a way to enable Trenz [TEG2000](https://shop.trenz-electronic.de/en/TEG2000-01-P001-FPGA-module-with-GateMate-A1-from-Cologne-Chip-16-MByte-QSPI-Flash-4-x-5-cm) participation in this future high-speed serial networking 

## 3-channel R-ladder 4-bit DAC
For the general (independent of Olimex) I/O Expander card, we are proposing to tripple-charge this rudimentary DAC by terminating it on the following three physical connectors, all in parallel:

**1) VGA**, in the likes of Basys3 and Olimex
> [image]

**2) 3.5mm TRS Audio plug**, for analog output to headphones w/o mic, and laptop/desktop sound inputs

**3.A) 3.5mm TRRS Audio/Video (AV) plug** -- While FPGA flexibility comes in handy for bridging the variety of connector pinouts, we wonder if the corresponding monitor and camcorders are still available, esp. given that the 2.5mm jack was also popular
>[image]

**3.B) 3.5mm TRRS Audio IO plug** -- Instead of AV, since we already have VGA option, the 2nd Ring could be used to bring in a Mic!
> [image]

chili-chips-ba, 2024-09-13:
The [SimpleIO](https://github.com/chili-chips-ba/openCologne/blob/main/0.doc/Intergalaktik/simpleIO.SCH.pdf) board is now available. 

[image]
[image]

closed 2024-09-27 by chili-chips-ba