chili-chips-ba/openCologne #18: How to 'report_timing' in CologneChip proprietary PNR tool?

chili-chips-ba, 2024-07-11:
Here is a good [question](https://discord.com/channels/1103723077814190092/1103728793245454488/1258374640070557717) from @TurboVega: 

> "... does anyone know how (command line option, maybe) to get more detailed information out of the P/R tool, such that we can determine exactly what "paths" are responsible for the maximum clock rates computed by the tool? In its output, it just labels the clocks with generated names, as it does when referring to various FPGA internal components, making it hard to know what Verilog source entities are involved. Knowing what parts of the source affect long paths helps in optimizing speed..."

pu-cc, 2024-07-11:
Due to mapping and various optimizations during implementation in P&R, it is not possible to keep all signals and names for cross-referencing. However, registers remain identical, and can be found in the `*.crf` output. The file is generated automatically after P&R if the `+crf` flag is set.

Here is an example: You find the critical path information with highlighted start and end in the P&R log:
[image]

The CPE names are made up of the component number (_a<number>) and the CPE part (/1) or (/2). In this example, the starting flip-flop has component number 110, part 2 (`110/2` or `_a110/OUT2`). You find it in the CRF file as follows:
[image]

In the post-synthesis netlist (`*_synth.v`), it has the instance name `_3208_`, and you will find the flip-flop with reference to `downsampler_inst.generalcounter[15]` in your code:
[image]

Similarly, we also find the target flip-flop (`100/1` or `_a100/OUT1`) in the CRF:
[image]

In the post-synthesis netlist, you find it as instance `_3199_`:
[image]

In order to optimizing your critical path, you could now examine the path between the`generalcounter[{15,6}]` registers in your code and optimize it if necessary.

TurboVega, 2024-07-11:
> The CPE names are made up of the component number (_a) and the CPE part

Thanks very much for your reply.

chili-chips-ba, 2024-07-11:
@pu-cc good tips 💯 

Still, how do we do random timing queries (such as `report_timing`) in the CologneChip framework?

Is there a document that describes scripts and procedures to use if they are not based on the de-facto industry standard SDC?!

chili-chips-ba, 2024-07-24:
@pu-cc - How do we go about specifying timing constraints for GateMate?
- primary clock period
- generated clocks (such as from internal PLL) or RTL dividers
- exceptions: MCP, FP
- I/O delays

chili-chips-ba, 2024-09-09:
@pu-cc , @DadoCCAG -- Your answers to the above questions have become uber-critical at this point!

We are seeing PicoRV32, which is the essential element of our [TetriSaraj](https://github.com/chili-chips-ba/openXC7-TetriSaraj) application, not work properly at 100MHz due to unreported timing issues. 

While we can blindly reduce its clock to 10MHz and pray that's slow enough for PicoRV32 to work fine without any constraints, that's hardly an approach that's satisfactory. Moreover, it is plain unacceptable for the commercial / pro projects.

TarikHamedovic, 2024-09-10:
I went through the GateMate documentation and found this line:

```
Furthermore, the netlist is passed to the Place & Route tool for architecture-specific im-
plementation and bitstream generation. A netlist converter generates a generic netlist
from the Yosys or legacy netlist. The first steps of Place & Route comprise procedures for
speed or area optimization before mapping. After placement and routing, the static tim-
ing analysis (STA) might lead to further optimization steps and makes the Place & Route
software an iterative process of constraint-driven re-placement and re-routing steps to
finally achieve user requirements.
```

In which it says that after P&R there is an STA, but looking through the pages 80-86 of the GateMate FPGA Datasheet there are no options to spec
ify a clock constraint as other FPGA vendors have. And also there is no mention of a clock constraint in their workflow diagram below.

chili-chips-ba, 2024-09-10:
... this calls for some questions: 
1) What criteria are used for `*constraints-driven placement*` and `*constraints-driven routing*` in the situation when even the elementary clock period cannot be specified?!
2) What's the scope of `*STA implementation step*` in this context, w/o timing constraints whatsoever?

pu-cc, 2024-09-11:
> Is it that you simply "feel comfortable" with 10MHz [...]

No, not at all. Let me briefly address the most important points:

Placement takes place using the quadratic placement algorithm. After all signals have been routed, p_r _always_ runs an STA. This can also be seen in the log file:

```
[...]
Static Timing Analysis

Skew violation report using only 80% delay of data path
[...]
```

STA takes the current placement as a basis and calculates the maximum achievable frequency for _all_ clocks, as I have shown in my first answer. Each clock reports a maximum clock frequency and it's critical path.

Moreover, STA checks for clock skew and applies measures to reduce it.

Once the STA has finished, it should be ensured that the timing for the clock specified in the report is achieved.

In my experiments, picorv and vexcrisv reached about 30-50 Mhz (worst corner).

chili-chips-ba, 2024-09-11:
@pu-cc given that the necessary timing information is available in the P_R database, what would it take to bring the flow from its current reactive* timing closure methodology up to something that at least on surface resembles the mainstream pro-active approach?! 

Here is an idea:

> 1) allow declaration of the basic `clock constraint` in the CCF
> 2) provide post-processing script that would extract all Fmax reports from the P_R log and compare them to the declared input clock frequencies, flagging violations when below, and displaying the extent of headroom when met
> 3) in the next phase, build on top of it to add support for `generated clocks`
> 4) eventually add ability to parse the database and support `report_timing` command

___
(*) the current P_R is apparently `not timing-driven`. We understand that the P_R is using _quadratic placement_ algorithm. 
 - Where can we find more about its `constraints-driven placement` and `constraints-driven routing` properties?
 - What exact constraints are tapped into to drive that process?

closed 2024-09-27 by chili-chips-ba