chili-chips-ba/openCologne #19: Issue with ILA using Olimex GateMate board

TarikHamedovic, 2024-07-13:
After doing all of the steps needed to configure the Olimex GateMate board with the ILA we get some errors which I will explain in this issue.

# First Error
The first error happens when trying to run the `ws2812_gol` example in `example_dut`. After running the following line
`python3 ILAcop.py config -vhd ..\example_dut\ws2812_gol\src -t ws2812_gol`

I get this error message:
`Error! No ccf file was found in the given folder!!`

We assume that the .ccf error is because of the fact that the file is in VHDL, and we haven't gotten the VHDL example for Blinky to work on our openCologne project. I would need to get into this further.

# Second Error
We then tried the Blinky example that has the Verilog and VHDL file present, but then went with the tried and tested Verilog route. Then when running the following line:

`python3 ILAcop.py config -vlog ../example_dut/blink/src/ -t blink`

We then get the following error:
```
Upload to FPGA Board...
Traceback (most recent call last):
  File "/home/user/FPGA/Boards/GateMate/gatemate_ila/app/ILAcop.py", line 269, in <module>
    ILA_user = RuntimeInteractionManager(int(float(ILA_config_instance.ILA_sampling_freq_MHz) * 1000000),
  File "/home/user/FPGA/Boards/GateMate/gatemate_ila/app/RuntimeInteractionManager.py", line 148, in __init__
    self.com = Communication(self.bytes_per_msg, self.analysis_frequency)
  File "/home/user/FPGA/Boards/GateMate/gatemate_ila/app/communication.py", line 62, in __init__
    self.dev.set_configuration()
  File "/home/user/.local/lib/python3.10/site-packages/usb/core.py", line 915, in set_configuration
    self._ctx.managed_set_configuration(self, configuration)
  File "/home/user/.local/lib/python3.10/site-packages/usb/core.py", line 113, in wrapper
    return f(self, *args, **kwargs)
  File "/home/user/.local/lib/python3.10/site-packages/usb/core.py", line 159, in managed_set_configuration
    self.backend.set_configuration(self.handle, cfg.bConfigurationValue)
  File "/home/user/.local/lib/python3.10/site-packages/usb/backend/libusb1.py", line 812, in set_configuration
    _check(self.lib.libusb_set_configuration(dev_handle.handle, config_value))
  File "/home/user/.local/lib/python3.10/site-packages/usb/backend/libusb1.py", line 604, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 16] Resource busy
```
We then fixed the following error by commenting out the following line of code from the  `app/communications.py`  file found on line 62 `self.dev.set_configuration()`.
You can see the solution in our [1.Blinky--ILA-Test](https://github.com/chili-chips-ba/openCologne/tree/main/1.Blinky--ILA-Test) folder. 
We don't know if this is the correct solution, but the ILA works afterwards.

# Third error
Also for the blinky example to work on Olimex the .ccf file needs to be changed because it is written officially for the GateMate Evaluation Board.
The file is found in `gatemate_ila/example_dut/blink/src/`
And the led Loc  parameter needs to be `IO_SB_B6` .

D4VF0, 2024-07-13:
Thank you for trying out the ILA on the Olimex board.

Regarding the first error, I believe it is due to the path not being found. Since the software first searches for the CCF file, it highlights this issue when the path is incorrect. The problem depends on the system where you are running the ILA. I noticed that you are using different slash and backslash formats in the Blink example and the WS2812 example. Please check this, or try using the absolute path to see if the specified path is accessible as stated. I hope this solves your problem. 

The second error, as indicated, is due to the RP2040 port currently being in use. This means that the USB was not properly released by the previously executed configuration. It is interesting that it works without calling `set_configuration`. It likely uses the existing connection instead of attempting to re-establish it. I am unable to reproduce this error myself. If the configuration on the FPGA was successful, you can also start the ILA with `python3 ILAcop.py start -s`, which will not reconfigure the FPGA. You can then reset the RP2040 using the reset button before starting, which should resolve the "port busy" issue. Thank you for pointing out this error. It requires further analysis.

Regarding Error No. 3, the Blinky example is specific to the eval board, so I wouldn't necessarily classify this as an error.

Thank you and best regards 
Dave

chili-chips-ba, 2024-07-20:
@TarikHamedovic do you also see that only 2nd error is keeping this track open?
@D4VF0 , any additional insights into: `... the second error ... requires further analysis.` 
We are looking for your and Tarik's go ahead to close this issue.

D4VF0, 2024-07-21:
Hey,

I have looked more closely at the issue you found and have found a bug  
in the dirtyJTAG firmware.  
https://github.com/phdussud/pico-dirtyJtag/issues/27
This bug has already been fixed and closed, so I think the ILA should  
now run without errors, so the issue can be closed.

Thank you and best regards

Dave

TarikHamedovic, 2024-07-25:
After downloading the latest version of the `gatemate_ila` and changing the `config.py` parameters to:
```
UPLOAD_FLAGS = ' -b olimex_gatemateevb '
CON_DEVICE = 'oli'
```
The `p_r` tool is: 
```
> p_r
GateMate (c) Place and Route
Version 4.2 (1 July 2024)
All Rights Reserved (c) Cologne Chip AG
```
The `openFPGALoader` version is: 
```
>openFPGALoader -V
openFPGALoader v0.12.1
```
The `yosys` version is: 
```
> yosys
Yosys 0.40+22 (git sha1 fa0c5c1d4, clang++ 14.0.0-1ubuntu1.1 -fPIC -Os)
```

We double checked the `udev rules` are correctly set, and the command we ran was from the `app` folder:
```
python3 ILAcop.py config -vlog ../example_dut/blink/src/ -t blink
```

We still got the same problem provided here:
```
Traceback (most recent call last):
  File "/home/user/gatemate_ila/app/ILAcop.py", line 267, in <module>
    if not ILA_config_instance.upload():
  File "/home/user/gatemate_ila/app/ILAConfig.py", line 1617, in upload
    dev.set_configuration()
  File "/home/user/.local/lib/python3.10/site-packages/usb/core.py", line 915, in set_configuration
    self._ctx.managed_set_configuration(self, configuration)
  File "/home/user/.local/lib/python3.10/site-packages/usb/core.py", line 113, in wrapper
    return f(self, *args, **kwargs)
  File "/home/user/.local/lib/python3.10/site-packages/usb/core.py", line 159, in managed_set_configuration
    self.backend.set_configuration(self.handle, cfg.bConfigurationValue)
  File "/home/user/.local/lib/python3.10/site-packages/usb/backend/libusb1.py", line 812, in set_configuration
    _check(self.lib.libusb_set_configuration(dev_handle.handle, config_value))
  File "/home/user/.local/lib/python3.10/site-packages/usb/backend/libusb1.py", line 604, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 16] Resource busy
```

Now commenting out the `set_configuration()` doesn't fix the problem.
We also tried pressing the `RP-BOOT1` button, but to no avail.

D4VF0, 2024-07-26:
Thanks for the detailed report. This seems to be a Linux issue. I have now added a routine to disconnect the kernel drivers from all interfaces of the RP2040 dirtyJTAG firmware if they are busy before the program tries to use them. My commit should fix this: [Commit 9544026](https://github.com/colognechip/gatemate_ila/commit/9544026fa48d164be4a83f7441523553b7f5cd25). 
Please use the RP-RST1 button to reset the RP2040 on the Olimex board.
Thanks for uncovering the issue.

chili-chips-ba, 2024-07-26:
@D4VF0 let's not close this track before Tarik and Ahmed validate this second fix. Upon confirmation, they will then close it.

D4VF0, 2024-07-26:
Please make sure to update the firmware of the RP2040. More information can be found at:
https://github.com/colognechip/gatemate_ila?tab=readme-ov-file#use-with-the-olimex-gatematea1-EVB

aimamovic6, 2024-07-26:
After updating the firmware and working with the latest commit I'm happy to report that the blinky example works like a charm, except that I always get the following errors:

- after the device is initially connected or RP-RST1 is pressed:

```
Error configuring the device
Kernel driver for interface 1 disconnected.
Kernel driver for interface 3 disconnected.

Upload to FPGA Board...
writeTDI: read: usb bulk read failed [Errno 110] Operation timed out
```
- every time after that it's just:
```
Upload to FPGA Board...
writeTDI: read: usb bulk read failed [Errno 110] Operation timed out
```

But, like I said, the LED is blinking as expected and I have no issues capturing the signals.

D4VF0, 2024-07-27:
Thanks for trying it out. This error occurs because a kernel driver is already attached to the RP2040. The program then runs an automatic error handling feature that I have built in. If the "Resource busy" error occurs and the configuration of the device fails, the program automatically detaches the kernel driver from all interfaces of the RP2040. Whenever you reconnect or reset the board, it reconnects, and the old kernel driver becomes active again. You can also use udev rules to unbind the kernel driver. This is a well-known issue for Linux: https://github.com/libusb/libusb/wiki/FAQ#can-i-run-libusb-application-on-linux-when-there-is-already-a-kernel-driver-attached-to-it 
Nice that it works now. I think the issue can be closed now.

closed 2024-07-27 by chili-chips-ba