chili-chips-ba/openCologne #20: Issue with inout port of constraint file

TarikHamedovic, 2024-07-14:
This problem was found primarily trying to make [Amaranth HLS](https://github.com/amaranth-lang/amaranth) work with CC GateMate. The issue is that Amaranth, based on the Python code of the module, creates the given constraint file but it doesn't have the option to get pin direction data because no other vendor uses pin direction declaration inside the constraint file. The declaration is done in RTL.

I tried getting over this problem by using `Pin_inout` generally. But when doing the simple blinky example in Amaranth doesn't work. But when manually setting the pin directions correctly to `Pin_in` or `Pin_out` where needed it works fine.

Then I tried a simple Verilog Blinky example where I change `Pin_out` of the LED on the Olimex GateMate board to `Pin_inout` and then the example doesn't work.

Why is that? Why doesn't it work and what changes in the post-placement/routing netlist or bitstream to make it not work?
What change in the netlist makes it work again?

pu-cc, 2024-07-14:
What exactly do you want to achieve by specifying an output port as `Pin_inout`?

`Pin_inout` defines a **bidirectional** pin, as described in the documentation and [here](https://colognechip.com/downloads/ccgm1a1-evb-master.ccf). In addition, your definition will at least lead to such a warning, as P&R cannot explicitly assign the pin.
[image]

`Pin_inout` works perfectly well with bidirectional pins, `CC_IOBUF`, to be more precise. #20 is therefore not an error. I will get back to you with a solution to the other discussion about the CCF format after consultation with my team.

goran-mahovlic, 2024-07-15:
>> What exactly do you want to achieve by specifying an output port as Pin_inout?

In my case I have tried to get PS2 working on Olimex board, and managed to get keyboard running as it only needs input data.

Then I have tried mouse core that needs inout out - changed in to inout in verilog and ccf, but did mouse did not work.

Then I tried with working keyboard code that is working if ccf is set as input, but it does not work if I just change pins to inout in ccf.

pu-cc, 2024-07-15:
Thank you @goran-mahovlic , can I find the mouse core in this repository? I'd like to take a look at it.

goran-mahovlic, 2024-07-15:
PS2 core is here - https://github.com/trabucayre/GateMate_demos/tree/main/colorBarVGAPS2

It is will create Keyboard example that stops working if pins in ccf are set inout https://github.com/trabucayre/GateMate_demos/blob/main/colorBarVGAPS2/colorBarVGAPS2.v#L7

pu-cc, 2024-07-19:
With [parameter `ps2core` being set to 2 ](https://github.com/trabucayre/GateMate_demos/blob/b9ffcae7c8c5935a702de6476c0c88c94c36b172/colorBarVGAPS2/colorBarVGAPS2.v#L7), we will instantiate the [ps2kbd module](https://github.com/trabucayre/fpgalibs/blob/4672011c87b7ea3c89218b7537bf5d90429b26a9/hdl/ps2kbd.v).

As you can see in the module ports, there is no `inout` port at all:
https://github.com/trabucayre/fpgalibs/blob/4672011c87b7ea3c89218b7537bf5d90429b26a9/hdl/ps2kbd.v#L7C1-L15C3

Yosys log confirms:
```
     CC_ADDF                        53
     CC_BUFG                         1
     CC_DFF                        122
     CC_IBUF                         4 // only input ...
     CC_LUT1                        10
     CC_LUT2                        74
     CC_LUT3                        23
     CC_LUT4                        39
     CC_MX4                         14
     CC_OBUF                        17 // ... and output buffers
     CC_PLL                          1
```

With parameter ps2core set to 0, however, we will describe the behavior of an IOBUF:

Yosys log confirms:
```
     CC_ADDF                        87
     CC_BUFG                         1
     CC_DFF                        177
     CC_IBUF                         2
     CC_IOBUF                        2 // here is our IOBUF
     CC_LUT1                        19
     CC_LUT2                       113
     CC_LUT3                        25
     CC_LUT4                       102
     CC_MX4                         14
     CC_OBUF                        17
     CC_PLL                          1
```

P&R will map these buffers into IOBUF cells.

Btw., with the latest P&R release, you can now specify the `NET` keyword instead of the pin direction to automatically determine the direction:
```
Net  "ps2clk" Loc = "IO_WB_A0";
Net  "ps2data" Loc = "IO_WB_B0";
```

https://colognechip.com/downloads/cc-toolchain-linux.tar.gz
https://colognechip.com/downloads/cc-toolchain-win.zip

TarikHamedovic, 2024-07-20:
With the new CC p_r update that introduced the `NET` keyword. The Amaranth Blinky example works immediately without any p_r errors. 

I am continuing the work to make Amaranth work on GateMate boards.

closed 2024-07-20 by TarikHamedovic