chili-chips-ba/openCologne #21: FATAL ERROR in GateMate p_r Tools for OPL3 FM Synthesizer

TarikHamedovic, 2024-07-20:
I encountered a FATAL ERROR while running the GateMate p_r tools in the [4.Advanced--4--Yamaha-OPL3-FM-Synth](https://github.com/chili-chips-ba/openCologne/tree/main/4.Advanced--4--Yamaha-OPL3-FM-Synth) example. This test was performed using the latest yosys and p_r tools.

As discussed previously [here](https://github.com/chili-chips-ba/openCologne/issues/3), SystemVerilog support in yosys is currently rudimentary. Without plugins, yosys struggles with synthesizing package imports in SystemVerilog. Synlig requires a specific yosys version, making newer versions incompatible. Other tools in the works include [SVase](https://github.com/pulp-platform/svase), [yosys_slang](https://github.com/povik/yosys-slang), and [sv2v](https://github.com/zachjs/sv2v).

We opted for sv2v since yosys handles Verilog files better. We converted all SystemVerilog files from the [opl3_fpga](https://github.com/gtaylormb/opl3_fpga) repository, which is now updated on our openCologne repository.

Running make impl in the [3.build](https://github.com/chili-chips-ba/openCologne/tree/main/4.Advanced--4--Yamaha-OPL3-FM-Synth/3.build) folder for the OPL3 FM Synthesizer resulted in the following output, with the error specified at the bottom:
```
GateMate (c) Place and Route
Version 4.2 (1 July 2024)
All Rights Reserved (c) Cologne Chip AG

Command line Options:
       h:  False
       i: net/opl3_synth.v
       o:   opl3
     mXA:   True
     mMA:   True
    Cinp:  False
   Coutp:  False
     cdf:   True
     pin:   True
     plc:   True
       s:   True
     cCP:  False
     gCP:  False
      AF:   True
      dC:   True
    uCIO:  False
      pB:       
      sp:  False
     crf:  False
       v:  False
      Gs:      0
      om:      3
      sf:    0.8
      dl:   True
      df:   True
   m_spi:      0
      pr:   True
     cgb:  False
    cgbc:  False
     cgo:      0
      tm:      3
      tp:      0
   s_spi:      0
     ics:  False
     lib:   ccag
     ccf: /home/user/openCologne/4.Advanced--4--Yamaha-OPL3-FM-Synth/3.build/../1.hw/constraints/blink.ccf

X_CONVERT Netlist Converter

Library Mode used: CCAG_GateMate
......................................
FATAL ERROR: (195004): Invalid assign statement
program finished with exit code: 10
make: *** [Makefile:165: impl] Error 1

```
Examining the net/opl3_synth.v file generated after synthesis, the error points to the following line:
```
assign _08870_[1] = \channels.ch_abcd_cnt_mem.bankb  ? _05638_ : _05637_;
```

If necessary, I can upload this net file to our GitHub repository as well. The synthesis completes with some warnings, which I'll address with Greg. However, I don't believe these warnings are related to the FATAL ERROR.

## How to replicate this error:
To replicate the error fully, you need the sv2v tool. Installation instructions are available on the [sv2v GitHub repository](https://github.com/zachjs/sv2v) or in our [openCologne/0.doc/1.README.Tool-Installs.txt](https://github.com/chili-chips-ba/openCologne/blob/main/0.doc/1.README.Tool-Installs.txt) file.

1. Run `make synth` to generate the Verilog files using sv2v and then perform yosys Verilog synthesis.
2.  After synthesis, run `make impl`.

For a faster process, you can skip the sv2v installation part since I already uploaded the converted Verilog files. For synthesis, run:

```
./create_rtl_filelist.sh -v
make synth_vlog
```

chili-chips-ba, 2024-07-22:
@pu-cc since this is apparently Yosys bug, could you please help us fully understand the relationship between CologneChip synthesis (which we believe is a Yosys fork) and main Yosys trunk?!

- How is the work between you and Yosys dev team split in terms of GateMate support, maintenance and upgrades?

TarikHamedovic, 2024-07-22:
After trying out the `p_r` process with your commit, the `make_impl` command works and there is no `assign error`.

chili-chips-ba, 2024-07-30:
@TarikHamedovic , is this one ripe for closure?
@pu-cc , we would still like to understand the split of work between you and Yosys dev team wrt GateMate support, maintenance and upgrades

TarikHamedovic, 2024-07-30:
On my side the commit fixed the problem.

closed 2024-07-30 by chili-chips-ba