chili-chips-ba/openCologne #22: GateMate simulation quirks and problems

chili-chips-ba, 2024-07-20:
Given [this](https://github.com/chili-chips-ba/openCologne/issues/17#issuecomment-2241072471), and [this](https://github.com/chili-chips-ba/openCologne/issues/14#issue-2342580901), and more of that in internal discussion, let's continue with simulation issues using this track.

chili-chips-ba, 2024-07-20:
@tarik-ibrahimovic, please continue the sim related discussion on this track
> such as: _"... on other projects even the post-synth sim doesn't work in Icarus, neither in Verilator, again the simulation doesn't advance. Again I stumbled upon a RTL-hardware inconsistencies which I wanted to take a closer look at but now is extremely challenging..."_

@aimamovic6 , the same goes for you sim struggles.

tarik-ibrahimovic, 2024-07-23:
We had a problem simulating the post-pnr netlist which contained `CC_PLL` module. @pu-cc has clarified the source of the issue [here](https://github.com/chili-chips-ba/openCologne/issues/17#issuecomment-2242928305). 

I just need to touch on the "post-pnr sim works in Verilator, doesn't in iverilog" mentioned by @pu-cc and me, for clarification:

- When you use the `CC_PLL` in the design neither Verilator nor iverilog can simulate post-pnr without editing the post-pnr netlist
- When there is no `CC_PLL` in the design iverilog can't simulate because it's still blocked by the cpelib.v model of `CC_PLL` but Verilator doesn't mind

This has been verified by removing  `define USE_PLL`  in the `cpelib.v` and running post-pnr sim in iverilog. It is implied that the new `CC_PLL` model blocks the post-pnr sim, whether it has the PLL in design or not. This is yet to be fixed.

pu-cc, 2024-07-23:
> When you use the CC_PLL in the design neither Verilator nor iverilog can simulate post-pnr without editing the post-pnr netlist

Exactly, that's why I suggested manually entering the `OUT_CLK` parameter as a real value. Next P&R update will pass these parameters from the synthesis netlist to the post-implementation netlist.

> When there is no CC_PLL in the design iverilog can't simulate because it's still blocked by the cpelib.v model of CC_PLL but Verilator doesn't mind

If no PLL is used, default parameter for `OUT_CLK` is zero. Please check the updated [cpelib.v](https://colognechip.com/downloads/cpelib.v); we've already updated it in the toolchain downloads.

chili-chips-ba, 2024-09-11:
@tarik-ibrahimovic please advise whether all PLL model questions are now resolved. If so, please close this issue.

closed 2024-09-11 by tarik-ibrahimovic