chili-chips-ba/openCologne #27: Error 10 on simple keyboard module

goran-mahovlic, 2024-07-30:
On apple_one I have ps2 keyboard module (parameter ps2core = 3) that is trowing error 10

Maybe someone can check here what is reason we get such an error - this is just PS2 module extracted from apple one so we can focus on non working part...

https://github.com/goran-mahovlic/GateMate_demos/tree/main/colorBarVGAPS2

By the way with new toolchain keyboard is now working with defined inout 

I still did not get any mouse parametar 0 or 1 to work, but there is no more errors...

chili-chips-ba, 2024-07-30:
@trcwm could you please help us understand the intent behind deliberate mix of blocking and non-blocking assignments in a clocked block.
> [image]

Also, state variable has 3 bits for only 4 states. 

@goran-mahovlic what test fixture are you using to validate the keyboard? Is there a wrapper RTL that `ps2keyboard.v` is set into? Have you checked whether clock was present and pinout correct?

@ADVIKBAHADUR , to learn the flow, try pushing _ps2keyboard.v_ through all three Linters: Verilator, Slang and Verible. Then use ILA to see what is going on inside...

ADVIKBAHADUR, 2024-07-31:
Upon linting on all three, there doesnt seem to be any errors, just a few warnings that dont seem to be anything major from what i can tell.

I ll share the all outputs for the warnings here if i have missed anything. (have not included some of the style warnings.)

`%Warning-UNUSED: ps2keyboard.v:39: Bits of signal are not used: 'rxshiftbuf'[0]
                                 : ... In instance ps2keyboard
    reg [10:0] rxshiftbuf;   
               ^~~~~~~~~~
                 ... Use "/* verilator lint_off UNUSED */" and lint_on around source to disable this message.
%Warning-UNUSED: ps2keyboard.v:49: Bits of signal are not used: 'ascii'[7]
                                 : ... In instance ps2keyboard
    reg [7:0]  ascii;        
               ^~~~~
%Warning-CASEINCOMPLETE: ps2keyboard.v:243: Case values incompletely covered (example pattern 0x0)
                                        case(rx)
                                        ^~~~
%Warning-CASEINCOMPLETE: ps2keyboard.v:179: Case values incompletely covered (example pattern 0x0)
                                        case(rx)
                                        ^~~~
%Warning-BLKSEQ: ps2keyboard.v:329: Blocking assignments (=) in sequential (flop or latch) block
                                  : ... Suggest delayed assignments (<=)
                            next_state = S_KEYNORMAL;
                                       ^
%Warning-SYNCASYNCNET: ps2keyboard.v:26: Signal flopped as both synchronous and async: 'rst'
                       debounce.v:44: ... Location of async usage
    if (rst)
        ^~~
                       ps2keyboard.v:65: ... Location of sync usage
        if (rst)
            ^~~
%Error: Exiting due to 6 warning(s)`

I will look further and update

chili-chips, 2024-08-01:
This is a good first step, designed primarily for you to familiarize with opensource RTL linters. Don't forget to pull-in `-Wall` switch. Before jumping to ILA, it'd be nice to also try this design in a functional sim...

pu-cc, 2024-08-01:
Before we get into a deeper analysis, we should first look at the P&R log.

```
FATAL ERROR: (5039): Invalid assign statement
program finished with exit code: 10
```

Same issue as in https://github.com/chili-chips-ba/openCologne/issues/21#issuecomment-2242973742.

In your specific case, @goran-mahovlic, it might be advisable to keep the `default` cases, such as:
https://github.com/goran-mahovlic/GateMate_demos/blob/bc013635505f0b1d16daf32b32af0a1e7390ade2/colorBarVGAPS2/ps2keyboard/ps2keyboard.v#L296-L303

goran-mahovlic, 2024-08-01:
@pu-cc  tnx!

Original code is here - https://github.com/lawrie/apple-one/blob/master/rtl/ps2keyboard/ps2keyboard.v

So I probably removed that default as I was trying to clear the error.

As there are already many boards that work with this code with opensource toolchain I as bit lost. 

https://github.com/lawrie/apple-one/tree/master/boards

And not only me as few people on Olimex Discord also lost a lot of time to understand what is wrong here.

Besides of keyboard, uart module on same project is failing to, but I will try over weekend with your yosys fork...

chili-chips, 2024-08-01:
... given that @tarik-ibrahimovic 's UART RTL design has also uncovered two issues with implementation flow, one in Synth, another in P_R (see [this](https://github.com/chili-chips-ba/openCologne/issues/17#issue-2354829763)), it is crucial to update to the latest CologneChip tool patches.

Let's recall that @ADVIKBAHADUR linting has also flagged missing `case default` clauses: _Warning-CASEINCOMPLETE: ps2keyboard.v:243,  Warning-CASEINCOMPLETE: ps2keyboard.v:179_. Perhaps even more importantly, Advik reported `Warning-BLKSEQ: ps2keyboard.v:329: Blocking assignments (=) in sequential (flop or latch) block`. More warnings are likely to pop up when this initial batch is cleared and `-Wall` switch also pulled in. 

While it seems that outdated tool flow is the main culprit, it would still be of interest to get @trcwm 's take on Lint report, along with corresponding RTL updates...

trcwm, 2024-08-02:
> @trcwm could you please help us understand the intent behind deliberate mix of blocking and non-blocking assignments in a clocked block.
> 

The blocking assignments are used as VHDL variables. Or well, they should be used that way. I haven't seen this code in ages but I know it worked on the various boards I tested it on. I see you're not using the original repo alangarf/apple-one. That is the official one.

chili-chips-ba, 2024-08-02:
@ADVIKBAHADUR it'd be interesting to examine the netlist in order to check whether Yosys inferred flops for `next_state`.
[image]

If so, all state transitions would be delayed by one clock. While probably not design intent, that would not cause problems in real life (since PS2 is incredibly slow). 

Given that we in this case don't see much value-add from the explicit tracking of `next_state`, one option would be to completely eliminate it, and use `cur_state <=` instead...

chili-chips-ba, 2024-09-07:
@goran-mahovlic , if you deem this issue resolved, please proceed with closing it.

closed 2024-09-13 by chili-chips-ba