chili-chips-ba/openCologne #31: No warning issued for unconstrained ports

tarik-ibrahimovic, 2024-08-20:
If declared top-module pins are not constrained in the `.ccf` file the Place and Route tool does not issue a warning, but it maps the IO to a tool-decided location.

pu-cc, 2024-08-20:
The `*.pin` file provides information about pre-placed and auto-placed IOs.

A warning for unused pins and a proper abort for missing CCF files are already implemented for the upcoming release. https://github.com/stnolting/neorv32/discussions/983#discussioncomment-10312885

pu-cc, 2024-09-03:
The latest release now issues a warning for unused pins and a proper abort for missing CCF files.

https://colognechip.com/downloads/cc-toolchain-linux.tar.gz
https://colognechip.com/downloads/cc-toolchain-win.zip

We can close this issue.

JulianKemmerer, 2024-09-13:
Hello there

When I run `p_r` with an empty .ccf file I dont get any warnings or messages about auto-placed pins
```
p_r -i top_265f_synth.v -o top_265f -ccf clocks.ccf &> top_265f.log
```
I get log like:
[top_265f_synth.v.txt](https://github.com/user-attachments/files/16987875/top_265f_synth.v.txt)
[clocks.ccf.txt](https://github.com/user-attachments/files/16987877/clocks.ccf.txt)
[top_265f.log](https://github.com/user-attachments/files/16987883/top_265f.log)

which has no warnings or messages about pins being auto placed.

I dont know what I did to not see the `is not positioned in CCF file and will be auto-placed` message I know I saw at least once before... maybe has to do with providing an empty .ccf file instead of a partial one?

pu-cc, 2024-09-13:
Thank you @JulianKemmerer, I have just looked into it.

Empty CCF files will be marked as "not ok", and then any CCF routines do not take effect. If at least one IO is constrained, it works. I have just corrected that. It will be included in the next update.

chili-chips-ba, 2024-09-27:
@pu-cc, is this update already out?