chili-chips-ba/openCologne #34: Out of context / Virtual Pin Support

JulianKemmerer, 2024-09-12:
Hi there,

I am using the Cologne Chip toolchain `p_r` for timing analysis with [PipelineC](https://github.com/JulianKemmerer/PipelineC). Specifically, what I am doing requires knowing the the FMAX of arbitrary HDL modules.

Being arbitrary internal HDL modules their IO ports are not meant to be mapped to physical FPGA pins. Other tools call this 'out of context'/ 'virtual pins'.

I currently supply an empty `.ccf` file in my attempt to do this (no .ccf file is error).

Are there additional flags that should be used?

For example I might see something like this in the log file
```
found components:
        CC_ADDF       129
        CC_BUFG         1
         CC_DFF       385
        CC_IBUF       257
        CC_OBUF       129
...
FATAL ERROR: Pin count to high:   Pins needed: 386, Pins available: 144
program finished with exit code: 3
```
(note: I dont see messages about pins being auto placed like I had once seen :shrug: 
```
is not positioned in CCF file and will be auto-placed
```
)

---

@TarikHamedovic kindly pointed me to the yosys flag `-noiopad` that says it is related to out of context style operation.

When trying that argument I see
```
found components:
        CC_ADDF       129
        CC_BUFG         1
         CC_DFF       385
...
MAPPER started

Map Adder
1 Adder chains mapped

During map of 0 CPEs 0 gates deleted!

64 CPEs replaced by double bit CPEs
#### Error: Exception Handler called. ExitCode: 112
Exception Class: ERangeError
```

Maybe this exception is buggy behavior or just improper tool use?

Thanks for your time!

chili-chips-ba, 2024-09-13:
**Re:** _"...I don't see messages about pins being auto placed like I had once seen..."_,  please check [this](https://github.com/chili-chips-ba/openCologne/issues/31). It sounds like you are saying that its fix had subsequently been annulled. If so, please reopen the issue.

@pu-cc , for context, _"...knowing the FMAX of arbitrary HDL modules..."_ is uber-critical for PipelineC auto-pipelining feature. `report_timing` from [this](https://github.com/chili-chips-ba/openCologne/issues/18#issuecomment-2342769246) discussion track would fit that bill...

JulianKemmerer, 2024-09-13:
> **Re:** _"...I don't see messages about pins being auto placed like I had once seen..."_, please check [this](https://github.com/chili-chips-ba/openCologne/issues/31). It sounds like you are saying that its fix had subsequently been annulled. If so, please reopen the issue.

Yeah does sound that way.

chili-chips-ba, 2024-09-27:
@pu-cc @JulianKemmerer, any further proposals on how to make progress on this track?

JulianKemmerer, 2024-09-27:
IIUC my current way of using the tool, with an empty .ccf file, will soon become an error? https://github.com/chili-chips-ba/openCologne/issues/31

From there - I suppose what I am asking for is some way to first, turn that error off, and second some way to not require auto-placement of pins. Maybe possible if the `-noiopad` yosys arg is used so no IOPADs get into the design. I only need timing from register to register. Timing for paths to/from IOPADs is not important (all IO is registered).

chili-chips-ba, 2025-07-12:
@mmicko any goodie you can share on this topic for the new opensource PnR flow for GateMate, that's based on nextpnr (aka Project Peppercorn)?!

chili-chips-ba, 2026-10-21:
Answered for the open flow: nextpnr --vopt allow-unconstrained places unconstrained I/O for out-of-context timing (see comment above). Reopen if PipelineC needs more.

closed 2026-10-21 by chili-chips-ba