chili-chips-ba/openCologne #36: P&R Error: ERangeError when connecting BUFG to user logic in GateMate FPGA

harunkovacevic, 2024-09-17:
The design includes a clock divider and attempts to use the CC_BUFG primitive to distribute a clock signal generated from user logic. However, when running place and route (P&R), I encountered the following error:

**#### Error: Exception Handler called. ExitCode: 112
Exception Class: ERangeError**

Here is the relevant Verilog code snippet:

```
wire clk_50mhz, lock;

pll pll_inst (
    .clock_in(CLK),        // 10 MHz input clock
    .rst_in(~reset),
    .clock_out(clk_50mhz), // 50 MHz, 0 deg output clock
    .locked(lock)
);

wire rst_s = ~lock;

// Divider %2
reg r_25MHz;
always @(posedge clk_50mhz or posedge rst_s) begin
    if (rst_s)
        r_25MHz <= 0;
    else
        r_25MHz <= ~r_25MHz;
end
```

```
// Attempt to use CC_BUFG for the divided clock signal
wire clk;
CC_BUFG pll_bufg (.I(r_25MHz), .O(clk));
```
I suspect that GateMate FPGA architecture does not allow connecting CC_BUFG primitive to signals generated by user logic (like the divided clock in this case), but instead restricts its use to dedicated clock resources.

In the original pll module, BUFG primitive is removed.

chili-chips-ba, 2024-09-18:
... @pu_cc the natural follow up questions are:
1) what's your recommended method for the generation of internal divided clocks, apart from PLL?
2) will the P_R tool auto-insert appropriate buffer to the internal "generated" clock nets, such as this one?

pu-cc, 2024-09-18:
> I suspect that GateMate FPGA architecture does not allow connecting CC_BUFG primitive to signals generated by user logic

No, that's not the case. @chili-chips-ba has already correctly referred to this in the datasheet.

Yosys automatically detects clock nets and inserts `CC_BUFG`, unless you specify `-noclkbuf`. It seems that this is a different problem. Where can I find the sources or a netlist to investigate this?

harunkovacevic, 2024-09-18:
Clone [TetriSaraj](https://github.com/chili-chips-ba/openCologne/tree/main/5.TetriSaraj) project. 
To reproduce error, after setting the **CC_TOOL** path in Makefile, use:
`cd 3.build`
`make all`.

I'm also sending P&R log file: [impl.log](https://github.com/user-attachments/files/17050006/impl.log)

When BUFG primitive is removed from _top.v_, P&R does not report an error, but the process finishes very quickly, and there is no information about the maximum frequency.
P&R log file for this case: [impl_wo_BUFG.log](https://github.com/user-attachments/files/17050057/impl.log)

chili-chips-ba, 2024-09-20:
Hi @pu-cc. This tool bug is blocking us. If fixing it would take time, any chance to share a proven example for generation of 25MHz clock from 10MHz input?! 

Have you tried running your [PicoSOC](https://github.com/pu-cc/picorv32/blob/gatemate-demo/picosoc/picosoc.v) on Olimex, using a CC_PLL-generated 25MHz or 50MHz clock?

pu-cc, 2024-09-20:
It is not a tool issue.

None of those signals is connected in your design. Please always check the synthesis warnings.

```
../1.hw/top.v:113: Warning: Identifier `\reset_n' is implicitly declared.
../1.hw/top.v:200: Warning: Identifier `\btnU_out' is implicitly declared.
../1.hw/top.v:200: Warning: Identifier `\btnR_out' is implicitly declared.
../1.hw/top.v:200: Warning: Identifier `\btnL_out' is implicitly declared.
../1.hw/top.v:200: Warning: Identifier `\btnD_out' is implicitly declared.
../1.hw/top.v:200: Warning: Identifier `\btnC_out' is implicitly declared.
```

Please add the `reset_n` to your input ports. Furthermore, please make sure to properly assign the buttons, e.g.:
```
assign buttons = { btnC, btnD, btnL, btnR, btnU}; // instead of *_out
```

> When BUFG primitive is removed from top.v, P&R does not report an error, but the process finishes very quickly, and there is no information about the maximum frequency.

If you remove it, please make sure to assign `r_25MHz` to `clk`. Otherwise, clock is not connected and the design is empty; check this in the synthesis logfile.

With those edits, it worked just fine here.

> any chance to share a proven example for generation of 25MHz clock from 10MHz input?!

Not sure if I misunderstand. You can use a PLL to generate a 25 MHz signal from a 10 MHz input?

closed 2024-09-21 by harunkovacevic

harunkovacevic, 2024-09-22:
The issue was related to `reset_n`. On the other hand, I would point out that the reporting from P&R tools was rather unclear in this particular case.

Thanks for helping solve the problem.

chili-chips-ba, 2024-09-22:
@pu-cc - Is there anything you can do to improve the P_R error reporting facility for this particular issue?  A more descriptive tool message would have preempted it altogether...

pu-cc, 2024-09-22:
The tool had already reported the problems.

```
../1.hw/top.v:113: Warning: Identifier `\reset_n' is implicitly declared.  <-
../1.hw/top.v:200: Warning: Identifier `\btnU_out' is implicitly declared.
../1.hw/top.v:200: Warning: Identifier `\btnR_out' is implicitly declared.
../1.hw/top.v:200: Warning: Identifier `\btnL_out' is implicitly declared.
../1.hw/top.v:200: Warning: Identifier `\btnD_out' is implicitly declared.
../1.hw/top.v:200: Warning: Identifier `\btnC_out' is implicitly declared.
```

> The issue was related to `reset_n`

As mentioned in https://github.com/chili-chips-ba/openCologne/issues/36#issuecomment-2362814935. The first warning refers to this problem.

chili-chips-ba, 2024-09-22:
... sure, but it's still odd that a Warning /* that leaves an impression of tool being on top of it*/ later on results in a hard crash with cryptic message:
> **#### Error: Exception Handler called. ExitCode: 112**
> **Exception Class: ERangeError**