chili-chips-ba/openCologne #37: L2T4 logic funcion gaps compared to standard LUT4?!

chili-chips-ba, 2024-09-27:
Is there an analysis of the percentage of 4-input logic functions that the GateMate L2T4 structure cannot implement?!

tarik-ibrahimovic, 2025-02-12:
I outlined the analysis of the differences between the structures [here](https://github.com/chili-chips-ba/openCologne/tree/main/8.StressTest#lut-tree-logic). @pu-cc if there's anything you'd like to add or modify so we can close this issue.

PythonLinks, 2025-02-12:
What a great article.  I have been wondering about these issues for a while.  So I linked to it from:
https://github.com/PythonLinks/awesome-gatemate

I will be including this material in a talk on "Why Gatemate", to be recorded and given next week.  

I now suspect that there are other great articles on this github repository.  Now I have to read all of them.

tarik-ibrahimovic, 2025-02-12:
@PythonLinks Glad to hear! Stay tuned for the `8.StressTest`, as there's a lot more coming in the following period. You might also find `8.StressTest/1.corescore_cc` useful for some initial analysis of the effect of LUT-trees on logic capacity. Where will the talk be held? Will it be available online?

PythonLinks, 2025-02-12:
Next week   17.02. -22.02, 2025, Radiona Zagreb is having a work shop on "A library for Embedded Debugging Tools. "
 https://github.com/hdl4fpga/hdl4fpga

My talk will be at 4pm.   I expect to post the recording to the two GateMate discord channels. 

 https://discord.gg/v5GcJa69
and
https://discord.gg/spJU3g3n

But frankly my expectation is that I will make some serious mistakes in the first versions of this talk, so I will be recording it again, and including people's comments and corrections. 

"I never do anything right the first time."

chili-chips-ba, 2025-02-12:
@TarikHamedovic , @harunkovacevic , @msultanovic and @nsokolovic7 will be there too!

@pu-cc are you coming to Zagreb next week to this event that @goran-mahovlic is putting together?

PythonLinks, 2025-02-12:
I would love to visit Zagreb, and bicycle around, but right now it is too cold.  I understand that you will have a similar event in a few months, so I will probably show up then. 

I live in Katowice Poland. 

I will be at the CERN 2025 FPGA conference.

goran-mahovlic, 2025-02-12:
Yes we have few things, first is this Saturday 15.02 we have Open Hardware meetup. 

It will be open and anyone can take stage and present open project in person.

Next we will have FPGA meetup on Wednesday 19.02. - I still do not have and schedule but as @PythonLinks already mentioned few people will join online. 

Also most people will be in Radiona  presenting FPGA project they work on, I will also try to record.

There is also two day workshop that will happen 20.02 and 21.02

https://radiona.org/radionica-hdl4fpga-na-ulx3s/

https://radiona.org/engineer-in-residence-miguel-angel-sagreras-argentina-hdl4fpga-repository/

chili-chips-ba, 2026-10-21:
Answered by the L2T4 analysis in 8.StressTest/README.md (https://github.com/chili-chips-ba/openCologne/tree/main/8.StressTest#lut-tree-logic---l2t4).

closed 2026-10-21 by chili-chips-ba