chili-chips-ba/openCologne #38: Constraints-driven P_R?!

chili-chips-ba, 2024-09-27:
GateMate P_R is apparently using  _quadratic placement_ algorithm.
- Where can we find more about GateMate proprietary `constraints-driven placement` and `constraints-driven routing`?
- What exact constraints are tapped into to drive that process?
- Any CCF examples you can share that make use of them?

> [image]
Also [see](https://github.com/chili-chips-ba/openCologne/issues/18#issuecomment-2344233355)

chili-chips-ba, 2025-07-12:
@mmicko, any references you can share (doc or example) on this topic for the new opensource PnR flow?!

mmicko, 2025-07-12:
Regarding documentation for nextpnr all is part od docs, and papers explaining algorithms are not ours but available and not hard to find. For code examples https://github.com/YosysHQ/prjpeppercorn-test-cases is best reference.

chili-chips-ba, 2025-07-15:
How about specifying timing constraints, such as clock period, false paths, multi-cycle paths, async clock groups, I/O delays, etc.?

chili-chips-ba, 2025-07-16:
Yes, it is a good idea to have that attribute for GateMate too! And to also positively report the I/O flops that, for whatever reason, could not be placed in the I/O.