chili-chips-ba/openCologne #39: Can the QSPI pins and ConfigFlash be repurposed for User App?!

chili-chips-ba, 2024-09-30:
Would it be possible to, upon loading FPGA bitfile from ConfigFlash:
  1) Take over the corresponding FPGA pins and use them for other things?
          - possibly even allowing user access to the hardened QSPI controller within FPGA
  2) Use the Flash for app data storage
  
GateMate datasheet does not seem to be listing those options.
> [image]

Yes, we love to be uber-efficient (don't call us stingy 😉) and get extra mileage from Olimex-provided components.
> [image]

pu-cc, 2024-09-30:
> Take over the corresponding FPGA pins and use them for other things?

Yes, that is f course possible. The data sheet does contain information about this on page 84, using the `uCIO` parameter.

[image]

Even the example CCF contains the config pins, which can all be used by setting the `uCIO` parameter.
https://colognechip.com/downloads/ccgm1a1-evb-master.ccf

> possibly even allowing user access to the hardened QSPI controller within FPGA

The hardened QSPI controller is used exclusively for configuration.

> Use the Flash for app data storage

This follows from 1. that this is of course possible, too.

chili-chips-ba, 2024-10-01:
Thank you, this is very informative. 

Closing this question with a note that it would help to add a reference or link to the `-uCIO` tool option in the Config section of the datasheet.

closed 2024-10-01 by chili-chips-ba