chili-chips-ba/openCologne #40: PCB PMOD Request: Realtime clock

chili-chips-ba, 2024-11-03:
ULX3S RTC feature is very cool and unique. But, since not needed for many use-cases, we believe it is better to offer it as an add-on PMOD card. 

As Digilent already has such a card:
  - https://digilent.com/reference/pmod/pmodrtcc/start
ours could also come with a relay that allows turning on/off a 2 Amp / 250V consumer. The availability of a power switch would open the RTC to a whole new set of apps without needing additional PMODs

goran-mahovlic, 2024-11-29:
I can add relay but maybe having 220V is not super safe options.
Maybe instead we could have sold state relays for some DC voltages - like 12V/24V/48V

Here are the options on the mouser for the solid state relays

[image]

chili-chips-ba, 2024-11-29:
We have a request from multiple lycée and high school embedded clubs for an `Intelligent School Bell` project 🛎️. As the school bells operate on 220V, the relay would allow us to build it without much additional ado. The relay itself provides isolation. The terminal screw-in block should also be on an isolated part of the PCB.

That said, now that the first intended app is known, put your thinking cap💡on, looking to address this need using the most appropriate means...

goran-mahovlic, 2024-11-29:
You know that emard was using 433MHz on board antenna for such purposes?

There are lots of cheap 433MHz relay modules on the market

https://robu.in/product/generic-12v-one-channel-rf-wireless-relay-module-with-remote-control/

and just wire from FPGA pin as 433MHz transmitter

There is also bit more expensive solution on 433MHz - https://botland.store/sonoff-wifi-intelligent-drivers/16199-sonoff-rf-r2-230v-relay-rf-433mhz-wifi-android-ios-switch-6920075775709.html

I also have LoRa PMOD - and there are some LoRa solutions that could be used.

That way you have 220V away for the board you are programming.

But I can also do one PMOD with 220V AC solid state - that is maybe an hour of work.

chili-chips-ba, 2024-11-29:
Granted, they don't really need an FPGA for this intelligent school bell. Any micro-controller would do. However, the idea is to make it simple for them to start porting their existing embedded projects to FPGA. RF would complicate that task. 

We therefore vote for PMOD with 220V AC solid state relay.

goran-mahovlic, 2024-11-29:
> Granted, they don't really need an FPGA for this intelligent school bell. Any micro-controller would do. However, the idea is to make it simple for them to start porting their existing embedded [...]

Good, I will put 2x solid state relay on PMOD - will try to find SMD version so that bottom side does not have pins you can touch.

I will put it on 4 layer board with 220V traces in the middle of the board. Maybe even add some holes for printed plastic cover over relays.

chili-chips-ba, 2024-11-29:
... you can also take clues from the RF solutions you referenced earlier in the thread. Essentially, if those guys can put their sensitive electronics next to power relays, so can we.
[image]

goran-mahovlic, 2024-12-05:
[image]

chili-chips-ba, 2024-12-05:
How about posting schematics, esp. related to relay drive circuit. What is the part on the bottom?

goran-mahovlic, 2024-12-05:
Schematic is up to date: 

https://github.com/intergalaktik/Extension_Boards_for_Olimex_GateMate/blob/main/tmrIO/tmrIO.pdf

Bottom part you see is Battery holder, as we want RTC to preserve time even if we power OFF board.

I am using 5V relay so we have FPGA_VCC to 5V STEP-UP relay has 40 mA coil current so it is not power hungry. 
It can handle -  250 VAC, 30 VDC  - 3A

https://eu.mouser.com/ProductDetail/Omron-Electronics/G5NB-1A-HA-DC5?qs=xZ%2FP%252Ba9zWqYlwXQmF0rNMA%3D%3D

I decided to go with only BJT NPN transistor that should be able to start relay even if we use 1.8V. 

https://eu.mouser.com/ProductDetail/Nexperia/MMBT3904215?qs=LOCUfHb8d9s%2FW6hzAJGDbw%3D%3D

[image]

But as there are many options I am also open to suggestions....

goran-mahovlic, 2025-01-09:
Board is assembled, relays are clicking, I2C is not tested ...

chili-chips-ba, 2025-01-25:
Do you have RTL for testing I2C?

goran-mahovlic, 2025-05-10:
Board is at ChiliChips - I will make another one and test when I get time...

https://github.com/intergalaktik/Extension_Boards_for_Olimex_GateMate/tree/main/tmrIO

closed 2025-05-10 by goran-mahovlic