chili-chips-ba/openCologne #41: PCB PMOD Request: Quadrature Encoder For Educational Use

chili-chips-ba, 2024-11-03:
Now that we have _SimpleIO-NextGen_ card with a cursor-type joystick, we'd also like to open this discussion thread for a **Quadrature Encoder** type of an input device. 

This principle is used in many apps, from ordinary mouse and iPod music players, to industrial and even military. It is therefore important to have a good, open demo hardware for teaching the students this important topic. 

- https://en.wikipedia.org/wiki/Incremental_encoder
- https://www.adafruit.com/product/5001

goran-mahovlic, 2024-11-05:
If it is for students then best approach might be to go with simple one like this one on digilent PMOD - but it is also available as PMOD and I do not like to copy paste same boards

Adafruit looks super nice, but it is more expensive it will not fit in PMOD standard as it has more legs - but we might be able to fit it if we combine it with GPIO expander IC #6

chili-chips-ba, 2024-11-06:
You can also look at things like this:
1) [Slide switch](https://www.lcsc.com/product-detail/Rotary-Wheel-Switches_XKB-Connection-TM-2024A_C561480.html)
2) [Rotary switch](https://www.lcsc.com/product-detail/Rotary-Encoders_HanElectricity-SIQ-02FVS3_C41430892.html)

Combining it into [GPIO Expander](https://github.com/chili-chips-ba/openCologne/issues/6#issuecomment-2457686562) is a good idea!

chili-chips-ba, 2024-11-06:
Closing this issue, as it is now merged into GPIO Expander.

closed 2024-11-06 by chili-chips-ba