chili-chips-ba/openCologne #42: PCB PMOD Request: An array of Open Drain power transistors

chili-chips-ba, 2024-11-03:
Automation, industrial control and robotics projects need to drive motors and actuators. FPGA outputs typically don't have enough current and voltage capability to drive them directly. 

While there are many application-specific ICs custom-tuned for a specific load, we are hereby seeking an array of generic power drivers for educational projects, which:
- students can directly drive from FPGA
- and externally connect into a topology of choice (H-Bridge, D-Class Amp, BLDC and AC motor driver, Servos, etc.).

That's to facilitate labs on this subject. While Digilent has a PMOD along these lines, ours should come with more channels and higher power capability:
 - https://digilent.com/reference/pmod/pmodoc1/start
 - https://digilent.com/reference/pmod/pmodod1/start
 
To get this discussion started, @aimamovic6 will follow up with a description of his external circuit for an FPGA-based D-Class Amp with Sigma-Delta DAC.

aimamovic6, 2024-11-04:
This [blog post](https://www.mathaelectronics.com/how-to-build-a-high-efficiency-class-d-audio-amplifier-using-mosfets/) describes a nice way to build a Class-D audio amplifier using MOSFETs that can be adapted. It could also be useful to have a pair of transistors with open Source and Drain pins to give the user a choice of configuration. To save pins, some transistors could have Source grounded and only Drain left for user to connect.     

I suppose also a potentiometer could be used to control the volume. 

[image]

chili-chips-ba, 2024-11-04:
... since we are not looking for high power outputs from this PMOD, one option would be to find a multi-channel IC that `integrates the pre-amp drivers with output transistors`. Something like an all integrated 6-to-12 channels, with 0.5-to-2 Watts per channel. 

Then, for apps that call for more power, the user could externally connect two or more channels in parallel...

goran-mahovlic, 2024-11-04:
It would be good if someone could explore what is already available on github as open project - with some open license, as if something similar is available it would be easy to redesign it to our needs...

chili-chips-ba, 2024-11-05:
Here are some good options for driver chips. The first two are especially interesting:

- **[TC4427](https://www.lcsc.com/datasheet/lcsc_datasheet_2410010303_Microchip-Tech-TC4427VOA_C636893.pdf)** - 18V/1.5A `3MHz`, 2xHalf-Bridge driver
- [XL293D](https://www.lcsc.com/datasheet/lcsc_datasheet_2403150220_XBLW-L293D-XBLW_C18723575.pdf) - 36V/1A  `500kHz`, 4xHalf-Bridge driver. The same as [SN754410](https://www.lcsc.com/datasheet/lcsc_datasheet_2303020200_Texas-Instruments-SN754410NEE4_C1556091.pdf)
- [L6230](https://www.lcsc.com/datasheet/lcsc_datasheet_2409302136_STMicroelectronics-L6230PD_C5267678.pdf) - 52V/2.8A, `100kHz` 3xHalf-Bridge for FOC, with a spare Comparator

[image]

[image]

[image]

goran-mahovlic, 2024-11-05:
We are often using L298D - it is widely available and easy connectable with few wires. Problem is that it is using 5V logic and we have at max 3.3V, same problem is with L293

From this list only TC4426 looks like it can have 3.3V for logic (minimum is 2.4V so that would also work on Olimex top connectors)

Maybe this one is it is not to week - TB6612FNG

That one is also standard on modules, and not super expensive

https://hr.mouser.com/ProductDetail/DFRobot/DRI0044?qs=F5EMLAvA7ID2YFYEixwScg%3D%3D

chili-chips-ba, 2024-11-05:
The **TB6612FNG** fits the bill 👍 It is also available on [LCSC](https://www.lcsc.com/product-detail/Brushed-DC-Motor-Drivers_TOSHIBA-TB6612FNG-C-8-EL_C141517.html)

chili-chips-ba, 2024-11-14:
We recommend employing two or even four of these chips, and so in a way that minimizes the use of FPGA pins, yet w/o limiting circuit flexibility. This could be accomplished by: 
1) connecting the same-kind inputs together
2) but also providing a DIP switch option to negate half of them, by using an on-board discrete logic inverter chip
3) individually bringing all outputs to a connector

This then allows externally building both parallel (for the current boost) and push-pull (for the voltage boost) structures from these uncommitted outputs. The drive would typically be configured for:
- in-phase (i.e. non-inverted) in the parallel, and
- out-of-phase (i.e. inverted) in the push-pull scenario.

[image]

chili-chips-ba, 2024-11-15:
@pu-cc any requests or suggestions from your side?

goran-mahovlic, 2024-11-26:
Something like this?

4 motors + 2 encoder inputs

[image]

[image]

chili-chips-ba, 2024-11-26:
The layout looks nice and symmetric.
- Any chance to add two more driver chips, with corresponding screw terminals on the sides? 
- What are the "encoder" positions for?

Could you please also share the schematic draft for a quick review.

goran-mahovlic, 2024-11-27:
If I add more chips it would be to big to fit 3 boards to Olimex. 

If we stay at 2 we can do this:

[image]

[image]

Encoders are quite useful for motor feedback - so I will leave them.

Initial schematics is here:
 
[pwrIO.pdf](https://github.com/user-attachments/files/17931059/pwrIO.pdf)

chili-chips-ba, 2024-11-27:
We love horizontal stacking options and flexibility that this proposal entails! We are fully sold on it 💯. 

Two follow up questions:

**1)___** 
The "encoder" position seems to be a mere connector for bringing-in the output of a motor quadrature encoder. For the complete picture, let's share here the datasheet of a typical motor with such encoder. 
- Can it be connected directly to FPGA, without level shifters, or any other kind of adaptation?

[image]

**2)___**
Are PWR_FLAGS simply PCB test points, i.e. slightly larger copper landing pads that can be used for probing?
[image]

goran-mahovlic, 2024-11-28:
1) - You are correct, voltage on PULLUPs and on PIN1 should be 2.5V - encoder should still work, but we will need to test it once the boards arrive...
Also those two pins can be used for some other type of feedback - fir example we could have connection to some I2C current measurement board to track motor current. You could also use HAL sensor here...

2) Those flags are just there so KiCad can know those lines are power lines, and not to complain in electrical rules check, So nothing special. But I can also add some testing points as those are really useful for tracking problems...

goran-mahovlic, 2024-11-28:
[image]

goran-mahovlic, 2025-05-10:
One board is at ChiliChips - I will assemble one more and test when I get more time. 

https://github.com/intergalaktik/Extension_Boards_for_Olimex_GateMate/tree/main/pwrIO

closed 2025-05-10 by goran-mahovlic