chili-chips-ba/openCologne #43: PCB PMOD Request: USB2.0 PHY

chili-chips-ba, 2024-11-03:
USB2.0 is affordable and ubiquitous, with peripherals ranging from HIDs to webcams, and more. However, FPGAs typically don't have a good way of direct interfacing to USB (with a notable exception of Gowin LittleBee lineup). This proposal is to bridge that gap. 

We have not seen a similar PMOD on the market. The reason is that it is hard to transfer 480Mbps over a PMOD. Should we manage to do it, we would directly play into the market segment occupied by `SPI Camera` modules like these:
- https://www.arducam.com/spi-arduino-camera

Our PMOD could be based on one of the following technologies:
- https://www.gowinsemi.com/en/market/featured_detail/13 -- LittleBee FPGA would make it easier to transport 480Mbps over PMOD
- https://verisilicon.com/en/IPPortfolio/USB2PHY
- https://www.ti.com/lit/ds/symlink/tusb1210.pdf

Granted, there are chips that convert USB to SPI, but they are limited to 12Mbps (aka "Full Speed), thus not suitable for cameras.

chili-chips-ba, 2024-11-05:
Here are a few more references to USB PHY examples for FPGA, that need only an external connector and a few resistors, such as found on ULX3S:
**1) https://github.com/WangXuan95/FPGA-USB-Device**
2) https://github.com/emard/had2019-playground/tree/master/cores/usb
3) https://github.com/emard/ulx3s-bin/tree/master/fpga/dfu

[image]

goran-mahovlic, 2024-11-25:
Could you please check if pins on  this connectors are differential. 

They are not routed as diff are but I think USB will still work if they can be used as diff pair...

[image]

chili-chips-ba, 2024-11-25:
@TarikHamedovic , @Tsvetanusunov any insights into this question?

goran-mahovlic, 2024-11-25:
Here is initial version - let me know if you see something that could be better

[image]

[image]

goran-mahovlic, 2024-12-05:
I have checked differential pairs and they are connected right on Olimex board, so we can use them as pairs!
But I have switched P and N on the connector side.

[image]

I do still did not find if A1 has dedicated clock input pins or any pin can do true diff input, for example on ECP5 you can only use fast clock input on dedicated clock input pins?

As I have created experimental DVI-OUT/DVI-IN board it would be great to find that info....

pu-cc, 2024-12-05:
> I do still did not find if A1 has dedicated clock input pins or any pin can do true diff input,

Dedicated clock input pins are:
* `IO_SB_A8`: `CLK0`
* `IO_SB_A7`: `CLK1`
* `IO_SB_A6`: `CLK2`
* `IO_SB_A5`: `CLK3`

You can use them all as differential inputs.

It is also possible to use _any_ GPIO as a clock input. The only thing to note here is that the signal must be routed via the routing structure to the entry point of the global clock mesh. These paths are longer than via the dedicated clock pins, which is why it is essential to pay attention to clock skew.

goran-mahovlic, 2024-12-05:
Perfect! Thanks @pu-cc

goran-mahovlic, 2024-12-05:
[image]

chili-chips-ba, 2024-12-05:
Please, also post link to the schematic, that we have it all in one place.

goran-mahovlic, 2025-05-10:
USB V1 had Voltage pins connected wrong - V2 is on order, I tested with wire fix and one USB keyboard was working - others did not but I think problem is not HW but timing related. 

https://github.com/intergalaktik/Extension_Boards_for_Olimex_GateMate/tree/main/USB

closed 2025-05-10 by goran-mahovlic