chili-chips-ba/openCologne #44: PCB PMOD Request: Olimex connector to dual PMOD

goran-mahovlic, 2024-11-29:
As you already used double PMOD - I think this could be useful.

It will also add 2X support for any other 3.3V PMOD additionally to the one we have on board.

It is also double the speed of the existing one so we could probably run some things that would not work on embedded one...

[image]

I am using LSF0108 https://www.ti.com/lit/ds/symlink/lsf0108.pdf?ts=1732867687007&ref_url=https%253A%252F%252Fhr.mouser.com%252F

[image]

chili-chips-ba, 2024-11-29:
The double PMOD is a good idea. Just for record, the `0.90" center-to-center` distance will be strictly adhered to, so that the 3rd party dual PMOD cards can be mechanically plugged in.
[image]

Is there a chance to beef up the 3.3V PMOD supply? 
The mere 300mA for two PMODs seems rather limiting. Could we at least put down two 2.5->3.3V converters, one for each PMOD?!
[image]

goran-mahovlic, 2024-11-29:
I think we are fine with first 

[image]

I will check for up converter chip that can go bit more up - but I as board itself is powered from 2.5V and I do not know the limit we need to be careful not to drain more then the board can take. 

I think we could use 500 mA Resettable Fuses - PPTC and bit better UP converter - if it is enough for two USB devices it should be good for most of the PMODs x2

If someone plugs everything and we have 4 connectors that is 2A on 2.5V I would say that board will struggle or die on that.

chili-chips-ba, 2024-11-29:
... it may also be worth reaching out to @TsvetanUsunov and solicit his opinion on the power that we can safely draw from his Olimex

goran-mahovlic, 2025-05-10:
Tested with LED matrice and it works as expected

https://github.com/intergalaktik/Extension_Boards_for_Olimex_GateMate/tree/main/dualTranslation

closed 2025-05-10 by goran-mahovlic

PythonLinks, 2025-10-19:
This is perhaps the most important board in the whole collection.   It would allow one to connect the 24 bit 1BitSquared  [DVI pmod](https://1bitsquared.de/products/pmod-digital-video-interface)  to the Olimex GateMate board. 

The first thing everyone needs is to see is the video output. And not just VGA.  

How many would a group of us have to order in order for you to do a production run of this board?

How much power does that chip take?  From  the [data sheet](https://www.ti.com/lit/ds/symlink/tfp410.pdf?ts=1760818373132&ref_url=https%253A%252F%252Fwww.mouser.jp%252F)

> IDD Normal power supply current Worst-case pattern(2) 200 250 mA

chili-chips-ba, 2025-10-19:
@goran-mahovlic,  please see above.