chili-chips-ba/openCologne #45: PCB PMOD Request: Olimex extension template

goran-mahovlic, 2024-12-02:
When creating similar boards multiple times it may be good to have template, so I have created one for extension boards...

[image]

[image]

[image]

chili-chips-ba, 2024-12-07:
@TarikHamedovic take note, templates are a very efficient method for creating variants of the PCB in a given form-factor.
- https://docs.kicad.org/doxygen/prj_tmp.html

goran-mahovlic, 2024-12-07:
@TarikHamedovic I would also suggest you to check this video - on my template I have used only few things like variables and page links, but some other tips could also be useful on more complex boards.

https://www.youtube.com/watch?v=_ZjyeltLMAg

As for KiCad - new features that arrive in v9 are super cool, and one is on v8 available with plugins.

https://youtu.be/DBfI2wwno5o?si=qu_Hq13piqefoLNL

Our (pwrIO) #42 and partially (tmrIO) #40 is done that way  so any change on one part will result in change on second one.

Then you can route first part and just replicate routing on all other parts with one click.

goran-mahovlic, 2025-05-10:
Template was really usable as it saves time it is available here: 

https://github.com/intergalaktik/Extension_Boards_for_Olimex_GateMate/tree/main/extension_template

closed 2025-05-10 by goran-mahovlic