chili-chips-ba/openCologne #46: PCB PMOD Request: Olimex connector to dual DVI-IN/DVI-OUT

goran-mahovlic, 2024-12-05:
ON ULX3S we have DVI extension that can do DVI IN/OUT.

We already know that DVI out is possible, but we are not sure if DVI IN is possible without special converter chip. 

So it worth of trying if we could replicate ULX3S results...

[image]

chili-chips-ba, 2024-12-05:
Great idea! We can entertain the RTL side of it in the follow up project.

goran-mahovlic, 2025-05-10:
DVI-OUT is tested and working - when he gets some time @mmicko will check if simple LVDS IN is working.

https://github.com/intergalaktik/Extension_Boards_for_Olimex_GateMate/tree/main/dvi

closed 2025-05-10 by goran-mahovlic