chili-chips-ba/openCologne #49: One and only EU

goran-mahovlic, 2024-12-12:
We are notified on discord that claim on this repository does not stand

> There is currently one and only one FPGA vendor in Europe -- CologneChip

https://nanoxplore.com/

I will also notify NLnet to do correction on project page...

chili-chips-ba, 2024-12-13:
Interestingly, the wording on [CologneChip](https://colognechip.com/programmable-logic/gatemate) webpage now does not include that claim either! 
- @pu-cc any additional comments?

pu-cc, 2024-12-13:
As far as I remember the Cologne Chip marketing used the term "The European FPGA".

We knew about it before. Their solutions are intended for space applications, i.e. their products are not available for conventional markets.

goran-mahovlic, 2025-05-10:
Fixed in README

closed 2025-05-10 by goran-mahovlic