chili-chips-ba/openCologne #50: Installation Error

udge2022, 2024-12-21:
when I follow the installation instructions I get an error message when installing via the pip command
`pip install --editable .[builtin-yosys]`

```
C:\Users\User\Software\cologne chip\amaranth>pip install --editable .[builtin-yosys]
Obtaining file:///C:/Users/User/Software/cologne%20chip/amaranth
  Installing build dependencies ... done
  Checking if build backend supports build_editable ... done
  Getting requirements to build editable ... done
  Preparing editable metadata (pyproject.toml) ... error
  error: subprocess-exited-with-error

  × Preparing editable metadata (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [31 lines of output]
      Traceback (most recent call last):
        File "C:\WPy64-31180\python-3.11.8.amd64\Lib\site-packages\pip\_vendor\pyproject_hooks\_in_process\_in_process.py", line 353, in <module>
          main()
        File "C:\WPy64-31180\python-3.11.8.amd64\Lib\site-packages\pip\_vendor\pyproject_hooks\_in_process\_in_process.py", line 335, in main
          json_out['return_val'] = hook(**hook_input['kwargs'])
                                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "C:\WPy64-31180\python-3.11.8.amd64\Lib\site-packages\pip\_vendor\pyproject_hooks\_in_process\_in_process.py", line 181, in prepare_metadata_for_build_editable
          return hook(metadata_directory, config_settings)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "C:\Users\User\AppData\Local\Temp\pip-build-env-2jkyr36g\overlay\Lib\site-packages\pdm\backend\__init__.py", line 90, in prepare_metadata_for_build_editable
          return builder.prepare_metadata(metadata_directory).name
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "C:\Users\User\AppData\Local\Temp\pip-build-env-2jkyr36g\overlay\Lib\site-packages\pdm\backend\wheel.py", line 102, in prepare_metadata
          self.initialize(context)
        File "C:\Users\User\AppData\Local\Temp\pip-build-env-2jkyr36g\overlay\Lib\site-packages\pdm\backend\base.py", line 194, in initialize
          self.call_hook("pdm_build_initialize", context)
        File "C:\Users\User\AppData\Local\Temp\pip-build-env-2jkyr36g\overlay\Lib\site-packages\pdm\backend\base.py", line 152, in call_hook
          getattr(hook, hook_name)(context, *args, **kwargs)
        File "C:\Users\User\AppData\Local\Temp\pip-build-env-2jkyr36g\overlay\Lib\site-packages\pdm\backend\hooks\version\__init__.py", line 53, in pdm_build_initialize
          metadata["version"] = getattr(self, f"resolve_version_from_{source}")(
                                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "C:\Users\User\AppData\Local\Temp\pip-build-env-2jkyr36g\overlay\Lib\site-packages\pdm\backend\hooks\version\__init__.py", line 91, in resolve_version_from_scm
          version = get_version_from_scm(
                    ^^^^^^^^^^^^^^^^^^^^^
        File "C:\Users\User\AppData\Local\Temp\pip-build-env-2jkyr36g\overlay\Lib\site-packages\pdm\backend\hooks\version\scm.py", line 348, in get_version_from_scm
          return version_formatter(version)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "C:\Users\User\Software\cologne chip\amaranth\pdm_build.py", line 8, in format_version
          major, minor, patch = (int(n) for n in str(version.version).split(".")[:3])
          ^^^^^^^^^^^^^^^^^^^
      ValueError: not enough values to unpack (expected 3, got 2)
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.
```

chili-chips-ba, 2024-12-21:
Please send the path to the installation instructions you are following.

udge2022, 2024-12-22:
Thank you very much for reporting so quickly.
I find the instruction [here](https://github.com/chili-chips-ba/openCologne/tree/main/1.Blinky--Verilog-VHDL-Python.Amaranth/4.amaranth)

udge2022, 2024-12-22:
Hi chili-chips-ba,

Thank you very much for reporting so quickly.
I found the instruction here.

openCologne/1.Blinky--Verilog-VHDL-Python.Amaranth/4.amaranth at main · chili-chips-ba/openCologne<https://github.com/chili-chips-ba/openCologne/tree/main/1.Blinky--Verilog-VHDL-Python.Amaranth/4.amaranth>
[https://repository-images.githubusercontent.com/798146281/c31af6c8-370b-449f-a9be-b52007122334]<https://github.com/chili-chips-ba/openCologne/tree/main/1.Blinky--Verilog-VHDL-Python.Amaranth/4.amaranth>
openCologne/1.Blinky--Verilog-VHDL-Python.Amaranth/4.amaranth at main · chili-chips-ba/openCologne<https://github.com/chili-chips-ba/openCologne/tree/main/1.Blinky--Verilog-VHDL-Python.Amaranth/4.amaranth>
Spicing up the first and (no longer) the only EU FPGA chip with a flashy new board, loaded with a suite of engaging demos and examples =&gt; https://www.chili-chips.xyz/open-cologne | Also see http...
github.com

chili-chips-ba, 2024-12-22:
@TarikHamedovic is our resident expert for Amaranth and will take a look at it...

TarikHamedovic, 2024-12-22:
Amaranth is still not fully supported for CologneChip, I still need to do some more steps for the pull request to be finished.
Amaranth versions change and that makes some problems. I will try to fix your report and get back to you.

closed 2025-07-12 by chili-chips-ba