chili-chips-ba/openCologne #52: LUTRAM stress test underutilization

tarik-ibrahimovic, 2025-02-14:
Already discussed [here](https://github.com/chili-chips-ba/openCologne/issues/28) running `8.StressTest/LUTRAM_Stress_test` causes the router to fail at approximately 10% CPE utilization. Recognizing the challenge posed by the wide output multiplexer routing, I modified the test to focus solely on memory capacity—using the `8.StressTest/LUTRAM_adjusted_stress_test`, where outputs are condensed to a single bit. Even with this adjustment, we haven't exceeded 58% utilization.

I'm trying to determine which aspect of the design is causing the constriction, but I'm struggling to find the crucial information. Specifically, I have two questions regarding the debugging process:

1.  How to extract information about the routing fails tracing it to the source from messages such as: `FATAL ERROR: 106 not routed!` or `CPE_OUTMUX congestion at Component 333 with 353`
2. How to view cdf files? Like in this [presentation](https://colognechip.com/wp-content/uploads/Novel-GateMate-FPGA-Architecture-FPL2021.pdf).

chili-chips-ba, 2025-07-12:
@swittlich any insights you can share on how the now opensource PnR (aka Project Peppercorn) fares in this case?

chili-chips-ba, 2026-10-21:
Re-run on PrjPeppercorn (Yosys 0.69, nextpnr-himbaechel 0.11.1-77, seed 1): LUTRAM_stress_test now routes up to 96 instances (67% of CPE_LT, 14 min). Legacy `p_r` failed at 24. 128 instances (86%) fail placement: the chip is full. LUTRAM still maps to CPE flip-flops, 160 per 16x10 instance. Details: https://github.com/chili-chips-ba/openCologne/tree/main/8.StressTest#prjpeppercorn-re-runs

closed 2026-10-21 by chili-chips-ba