chili-chips-ba/openCologne #54: YOSYS synth_gatemate netlist not compatible with p_r

tarik-ibrahimovic, 2025-02-14:
This issue occurs in `8.StressTest/1.corescore_cc`. FuseSoC is integrating yosys+p_r as `gatemate` toolchain, however a seamless build experience isn't there due to yosys generating some lines which the p_r is failing to read properly.

Specifically, the problematic code is an automatically generated initial block in the netlist, example shown below. Removing this initial block solves the error, but it has to be done manually because it's in the middle of what FuseSoC defined as a toolchain, and can't be manipulated easily through  scripts. To recreate this issue run the provided build steps in `8.StressTest/1.corescore_cc`. 

At this point, I have not found any tool options to bypass this behavior. If no alternative solution is available we might offload this issue to yosys.
```
 initial begin
    if (1'h1) begin
      $write("Preloading $paramod$76ca5ad6cddf78bc518e7753a546bede55c5cafa\\serving_ram from %s\n", $unsigned(88'h636f72655f34362e686578));
    end
    if (1'h1) begin
      $write("Preloading $paramod$ea5542b74da5d3df7ad0239dce78e365096fcc62\\serving_ram from %s\n", $unsigned(88'h636f72655f33342e686578));
    end
    if (1'h1) begin
      $write("Preloading $paramod$f9cf07300b5f1f0ca052c9602528ae8d7d785ee0\\serving_ram from %s\n",
...
```

chili-chips-ba, 2025-02-14:
Hi @pu-cc any tips / suggestions from your side?
Hi @tarik-ibrahimovic, please open support questions for both Yosys and Fusesoc, linking in this issue.

tarik-ibrahimovic, 2025-02-14:
After reporting a yosys [issue](https://github.com/YosysHQ/yosys/issues/4905#issuecomment-2660348432) I found the error to be originating from a piece of Serv's [code](https://github.com/olofk/serv/blob/main/serving/serving_ram.v#L42)

This turns out to be a recognized issue, and has recently broken OpenLane ASIC [flow (https://github.com/YosysHQ/yosys/issues/4122).

chili-chips-ba, 2025-02-15:
Good catch! 
  You may also want to report to Serv developers.

tarik-ibrahimovic, 2025-02-16:
I tried working around the issue through a custom tcl script for yosys but found [issues](https://github.com/olofk/fusesoc/issues/732) with fusesoc.

pu-cc, 2025-03-05:
Can you give this a try?

https://colognechip.com/downloads/testing/cc-toolchain-linux.tar.gz
https://colognechip.com/downloads/testing/cc-toolchain-windows.zip

This should solve any issues with the `initial` blocks.

tarik-ibrahimovic, 2025-03-05:
This fixes it.

closed 2025-03-05 by tarik-ibrahimovic