chili-chips-ba/openCologne #55: Olimex board overheating

tarik-ibrahimovic, 2025-02-19:
While running [fpga_torture](https://github.com/chili-chips-ba/openCologne/tree/main/8.StressTest/5.fpga_torture) on an Olimex FPGA board, I observed an unexpected overheating issue. The design utilized 96% of the FPGA fabric at 120 MHz (with an Fmax of 137 MHz). During testing, the PLL Vdd supply line exhibited significant overheating, reaching a measured hotspot of 70°C before I halted the test for safety reasons. Given the visible board damage, the actual temperature may have been even higher.

The test was conducted under a 1.1V core voltage, with no external loads apart from an LED, ensuring that the FPGA was only powering its internal logic. This raises two key questions:

1. Is the Olimex board power dissipation limited? Could the voltage regulators or PCB thermal design be insufficient for high-utilization scenarios?
2. Were sufficient validation tests conducted to confirm the board’s capability under sustained high-load conditions?

Further investigation is needed to determine whether this issue stems from board limitations, inadequate cooling, or an unforeseen FPGA power draw characteristic.

[image]
[image]

chili-chips-ba, 2025-02-19:
@pu-cc a power estimation tool for GateMate was mentioned in the course of our earlier discussions. 

This test case seems like a perfect scenario to put it to good use, looking to assess FPGA power consumption. Could you please help us bring GateMate Power Estimation methodology into Tarik's `fpga_torture` play?!

chili-chips-ba, 2025-02-19:
@TsvetanUsunov any insights you'd like to share on this question from the board perspective?

pu-cc, 2025-02-19:
@tarik-ibrahimovic In general, leakage current increases with increasing temperature. Consequently, an increasing current is to be expected. We have documented the leakage current for the _core_ in our data sheet up to 125°C. That is ~500mA. For the `VDD_PLL` domain, however, this should be significantly lower.

Do you have a chance to measure the current directly at the DC-DC regulator output of `VDD_CORE`, i.e. U15? For example, you could assemble a shunt at `VDD_CORE_E1`.

Did you modify the board yourself? `C40` has a different component size on my board.

pu-cc, 2025-02-19:
@tarik-ibrahimovic Btw., your thermal camera seems to have an offset? The hotspots should be on the marked spots, not next to them. 

[image]

chili-chips-ba, 2025-02-22:
... here is a potentially related [datapoint](https://github.com/stnolting/neorv32/discussions/983) reported by @stnolting

[image]

pu-cc, 2025-02-23:
> ... here is a potentially related [datapoint](https://github.com/stnolting/neorv32/discussions/983) reported by [@stnolting](https://github.com/stnolting)
> 
> [image]

The incorrect resistance value of the filter - that was used at the time - has nothing to do with this issue. As you have already correctly referenced in the screenshot, the excessive ripple led to non-locking PLLs.

Any updates on the power consumption?
Any updates on the obvious offset with your camera?

Again, this does not look like a board damage due to high temperature; it looks like as if someone manually soldered `C40`. You can even see Flux next to it? We already run this test at 150 MHz at 1.1V. 

[image]
(see https://colognechip.com/docs/ds1001-gatemate1-datasheet-latest.pdf)

chili-chips-ba, 2025-02-23:
... while side-tracking from the main discussion topic of this thread, here is a relevant schematic excerpt.

[image]

It is interesting that the C40 is used in parallel with 2 other caps, within a larger π filter, which apparently was not sufficient w/o a very particular C40 component 🤔

pu-cc, 2025-02-24:
@chili-chips-ba How is this relevant? @tarik-ibrahimovic did not report any PLL locking issues.

chili-chips-ba, 2025-02-24:
... let's try to connect the dots:

1) Zooming into @tarik-ibrahimovic's picture, we can observe that the R5 is visibly charred.
>[image]

2) The schematic clearly shows that the PLL supply current flows trough the R5.

3) The region of high heat in @tarik-ibrahimovic's thermal image is centered on FPGA, but offset to the side, and towards the R5.

4) The flux melts for any source of heat, be it a controlled soldering process, or out of control resistive barbeque 🔥 ☁

chili-chips-ba, 2025-02-24:
Regarding **150MHz** Fmax reported in GateMate [datasheet](https://colognechip.com/docs/ds1001-gatemate1-datasheet-latest.pdf) for the @stnolting `fpga_torture`
>"...🔥Technology-agnostic FPGA stress-test: maximum logic utilization and high dynamic power consumption..."

it is not clear what `sample size` was used to derive that officially published number. Tarik's sample size is undoubtedly one.
[image]

The datasheet diagram is implying that 150MHz was attainable even with 0.9V supply, whereas Tarik's testing has hit the wall at the mere **137MHz**, and for an elevated-to-the-max 1.1V core supply voltage. Additionally, while Tarik's testing has maxed out at the (very impressive) **96%** utilization, the GateMate datasheet is boasting **100%** utilization number.
- Is Tarik missing something important that contributed to these discrepancies?!

While we understand that GateMate is offered in only one _speed grade_, it'd be interesting to also understand how the inevitable manufacturing **P**rocess variations are handled by CologneChip in both tools and physical devices. One simple (though rather inefficient) technique would be to use the lowest common denominator in all parametric specs...

pu-cc, 2025-02-24:
> If you still have doubts, I found an image of the same board before the damage occurred.

This image does not show the level of detail to recognize any difference. There are also brown spots from the flux; again, this is an indication that soldering was done manually and the flux is a residue which cannot be caused by simply heating of the tracks.

Flux liquefies around 150°C. This also contradicts your measurements.

> 1. [...] we can observe that the R5 looks slightly different from the R2 [...]

What you see on R5 is flux. From the incorrectly displayed thermal image it should also now be clear that the resistor has not become hot, more on this in 2:

> 2. [...] schematic clearly shows that the PLL supply current flows trough the R5 [...]

It is an RC filter, of course it does. However, this is the exclusive supply of the PLL DCOs (also labeled as `VDD_PLL`). Power consumption of a PLL DCO is *max* 0.7 mW, unattached of the utilization in the core (labeled as `VDD_CORE`). 0.7mW is *far* below the current that can flow through resistor `R5`. **It cannot lead to such heat development either.**

> 3. The region of high heat in @tarik-ibrahimovic's thermal image is indeed not centered on FPGA, but instead offset to the side, and towards the R5.

Yes, and that is very wrong. The camera is set up incorrectly. You can _clearly_ see the rectangular heatspreader from the FPGA housing *next* to the actual image. Same issue with the coil and the voltage regulator. The thermal image shows the heat *next* to them; **the camera does not provide a correct image**.

> 4. The flux melts for any source of heat, be it a controlled soldering process, or an out of control resistive barbeque🔥

This can be right, of course, but we did not see 150°C to liquefy the flux.

@tarik-ibrahimovic Thank you for taking the shunt measurements. Looking forward to seeing the results. Is your board still working? Is the PLL locking? The title of the issue is misleading, as the GateMate does not overheat.

In order to make progress, I will now repeat this test on our Olimex board.

chili-chips-ba, 2025-02-24:
@pu-cc that's an excellent idea! 
  Please also check Fmax (for 1.1V core voltage) and utilization numbers.

chili-chips-ba, 2025-02-24:
While the title of this issue references `Olimex`, not `GateMate`, please recommend a title that would be a better fit. 

Our boards were acquired from Olimex almost a year ago, and all are in the **RevA** lineage.

While it is totally unexpected to see Fmax incease with temperature (it is usually the other way around), @pu-cc results are so different from the ones reported by @tarik-ibrahimovic that something else might be in the puzzle. Tarik is redoing his experiments in order to get to the root of this enigma...

tarik-ibrahimovic, 2025-03-01:
I've found that the camera truly did have an offset and the **temperatures are well within the allowed range**. Also, I can confirm the Fmax figures for `-tm {-3,-2,-1}`, and I'm updating the results on our repository. We didn't get the 100% utilization numbers but topped out at 96.2%, but that is a topic for another discussion.

closed 2025-03-01 by tarik-ibrahimovic