chili-chips-ba/openCologne #56: Math Stress Test

PythonLinks, 2025-02-20:
There have been a number of stress tests which show the GateMate performing poorly compared to other more conservative architectures.  It would be great to show some cases where this innovative architecture performs better than competing architectures.  At the meeting yesterday it was suggested that this device is particularly good at math.  That makes sense.  Each of the 20K CPEs can be a two bit adder or multiplier.   And the adders are connected both horizontally and vertically.  One competing chip with 17K LUTs has 27 * 18 bit multipliers meaning 112 x 9 bit multipliers, or 10*36 bit multipliers.

I calculate 47 CPEs per 8 bit multiplier on the Gatemate.  Meaning it could maybe do about 400 * 8 bit multipliers. 

[image]

where GateMate would really look good would be 10 or 20 bit multipliers.  The chip mentioned above would have to use 18 it multipliers to provide a 10 bit multipler, and it only has 27 of those.  

GateMate would also look great as a 20 bit multiplier.  the chip mentioned above would have to use 36 bit multipliers, and only has 10 of those.  

The speaker mentioned an FFT as a good math test case.   A vector processor would be another good case.  Here are my generic slides on the topic. 
https://pythonlinks.info/presentations/vectors/Cern2025Vectors.pdf

And of course a Gaussian Logarithm processor, using look up tables and interpolation, would be a killer technology on this device, sadly no one is interested. 

I am curious what the designers of this chip were thinking when they created it?  Why did they choose LUT-2 trees?   

I am curious if there are any other applications perfectly suited to this device?

And when doing stress testing, the natural thing to do is to compare it to other devices which a similar number of LUTs.  A better comparison would be to mention the price of competing devices.  The market cares not about LUTs, but about Price/Performance.

tarik-ibrahimovic, 2025-02-20:
> There have been a number of stress tests which show the GateMate performing poorly compared to other more conservative architectures. It would be great to show some cases where this innovative architecture performs better than competing architectures. 

There's something coming into play in-between the RTL and FPGA architecture: **the tools**. While there are no direct apples to apples comparison of the synthesis and pnr tools contrasting the proprietary toolchains and yosys+(nextpnr/p_r), still there is a strong suspicion the current open-source toolchains are significantly behind in optimization capabilities.

Take for example that CologneChip has to do a [translation step](https://github.com/chili-chips-ba/openCologne/issues/28#issuecomment-2353907213) from the yosys netlist which implements logic in LUT4s to L2Tx primitives, which the p_r understands: 
> About the LUT-tree itself. We are currently using yosys/abc to map combinatorial logic into typical LUT4. P&R analyzes it and maps it into the LUT-trees. This is certainly not the best way to go 

Also, have in mind that some of these tests were suggested by CologneChip [themselves](https://github.com/chili-chips-ba/openCologne/issues/28#issuecomment-2354689482).

At this stage, it remains uncertain whether the performance bottleneck stems from the tools or the architecture itself. More testing is planned once nextpnr support is available, but for now, we can only rely on the currently observed performance figures.

Mentioned math capabilities are there just to say that even though there are logic deficiencies we're not lacking in neither math, nor multiplexing, which are crucial.

chili-chips-ba, 2025-05-13:
Hi @swittlich any update on your comprehensive Math Stress / DSP test for GateMate?

chili-chips-ba, 2026-10-21:
Thanks for the benchmarks and the discussion. These numbers come from the legacy p_r flow, which we no longer use: openCologne now builds with the open Peppercorn flow (Yosys + nextpnr-himbaechel + gmpack). Closing; please open a new issue with Peppercorn results if the question still stands.

closed 2026-10-21 by chili-chips-ba