chili-chips-ba/openCologne #57: Better Documentation

PythonLinks, 2025-02-21:
Personally I think it is brilliant that GateMate actually has multiplexers.  Using 31 traditional LUT-4s to multiplex 32 bits seems quite nuts to me. 11 CPEs working as MUX-4s seems much better.  But then with GateMate, I get lost because of the lack of detailed information.   Take this image for example.  

[image]

 There is some hidden magic in the “Logic Lut Tree”.  If it were just a LUT tree, it would make sense to me.  But sometimes I think it is used as a multiplexer.    And sometimes parts of the LUT Tree are used for the adders and multipliers.    I suspect that there is some brilliance going on here, where the same circuits can be used in vastly different ways.   But I am stupid.  I cannot quite see it.  If you could do a diagram of the entire CPE, and then highlight the parts that are used in each case that would be brilliant.  Who knows, maybe some of us would come up with other ways to use the pieces together.    

Same thing holds for networking nodes.  It is not clear to me where the bit flips occur.  I am not able to reason about why you cannot do 6 input muxes.    Maybe I just need to read that section more carefully.   

And then with the CCL2T5, I am again confused.  Is this really as good as a regular Lut4?  Please persuade me.  Is it different? Then please tell me how it is different.  

From a sales and marketing perspective it is really important to be clear about this stuff.  Many developers will put up with lots of small irritating problems if we are convinced that you have a great idea here.  I suspect that there is some brilliance in this design, but I am not entirely convinced.    

Please convince me.

chili-chips-ba, 2025-02-21:
Let's try to, with CologneChip help, prove the _brilliance_ point, and construct a test that would bring to life GateMate hidden treasures! 

Such test can be `fully customized for GateMate, looking to squeeze the last drop of its unique advantages`. It would be designed and fine-tuned to allow GateMate to showcase its best face. It can tap into the 8-input logic capability, muxing riches, math functions, or whatever else @pu-cc and we collectively believe would make a difference. 

We would then expose a couple of the mainstream FPGAs in the 20K LUT category to that exact same test that was custom-constructed for GateMate...

TurboVega, 2025-02-21:
I.e., stack the deck? Sorry, had to giggle.

PythonLinks, 2025-02-21:
Rather than stacking the deck, it is more about identifying the market niche for this interesting device. 

Please everyone contribute your ideas to this project.  Often one person's crazy idea will trigger a very reasonable suggestion from another person.  I will do mine for Math.   

To save energy biological systems use logarithmic representation.  In biological systems many senses use a logarithmic scale. We can hear differences of 1 decibel in amplitude, that corresponds to a sound 1.25 times louder. In frequency, an octave is divided into 12 notes, and each note is divided into 100 cents. For frequencies above 5Khz, we can
distinguish 10 cents, or 1/120 of the signal, or a little less than a 1% change in frequency. For light we can distinguish a change of 10% in amplitude. Of course color uses different 3 different sensors and so works differently.  When the just noticeable difference is proportional to the signal size, that implies a logarithmic number system. 

Without going into detail about how to use Gaussian Logarithms (also called fixed point logarithms) for doing multiplication, division, **addition and subtraction**, I think that this device is great for it.   Basically one needs to do table lookup, with some interpolation.  Say 12 bit logarithmic math, just a little bit larger than the hard core 9 bit adders on other devices.   

Any other suggestions?  I am particularly curious about  how one would use  8 bit LUT trees.

TurboVega, 2025-02-21:
What about vector math in a 3D pipeline? I ported Pingo into Agon, in C, of
course. It ran okay, but with no real hardware help, the frame rates were
minimal. And that board only has 64 colors. GM has 4096 colors. Much better
possibilities.

chili-chips-ba, 2025-02-21:
... it would also be very interesting and super-cool to try @JulianKemmerer's [PipelineC](https://www.youtube.com/watch?v=F8jlJapQbFY) HLS with GateMate. 

For that, CologneChip needs to provide proper timing models. @pu-cc may have updates on that front...

PythonLinks, 2025-02-22:
## RISC-V Stress Tests 

From my work on very small soft core stack machines, I learned that RISC-V is grossly inefficient on 4 input LUT devices.  You can watch the video here: (starts at 2:40).
https://www.youtube.com/embed/ZokT2tEiXSA

To multiplex 32 bits on a 4 input LUT device requires 31 4-LUTs.  To multiplex 32 x 32 bit registers requires 996 4-LUTs.  To do 2 arguments requires 1988 4-LUTs.  In contrast on GateMate, multiplexing 32 bits would be 11 CPEs.  352 CPEs for 32x32 registers.  If they can get the 8Mux to work, that would fall to 5 CPEs. or 160 total. Of course to that one has to add the cost of 32 bit multiplies.  I estimate 160 CPEs.  I could calculate it exactly.  In the video I did calculate it for 8x8 multiplies.   But overall a fraction, about 1/4, of what is required for RISC-V on 4-LUTs.   Plus this device is less expensive on Digikey. 

Instead what many RISC-V designs do is to store the registers in BRAMs, but then that requires more clock cycles to access, or a larger pipelined device.  In either case RISC-V is a killer app for GateMate. 

I actually built an "[Awesome RISC-V Soft Cores](https://github.com/PythonLinks/awesome-risc-v-soft-cores)" page.   The next step is to figure out which ones use proper registers, and which ones use BRAMs for registers.  One person said that many have a switch to choose.  And then that will make great marketing material for GateMate.  "Awesome RISC-V Soft Cores is a directory of the best soft core RISC-V machines."  And when reading it people would find out that they all run best on GateMate.  Based on the Digikey cost of the part, and the percentage occupied by RISC-V, one could give the cost of the different soft cores on the different chips.  A very powerful marketing message. 

> how do you feel about stacking some logarithms into a GateMate-optimized test deck, written in  Verilog.  

Generally that is what I will be doing.  It is where my strength lies. (Math on FPGAs).  But I think it is a terrible market niche.  Almost nobody thinks in terms of logarithms nowadays.  (I used to even have log and log-log graph paper in my filing cabinet). There is no money to be made there.   

Where is the money?
 
What I should be doing is porting RISC-V soft cores to GateMate, and comparing the cost of each soft core (percentage of resources * chip cost)  on GateMate versus on other devices.   There is a real business case to be made for GateMate.   Now that Trump is ditching Ukraine, the risk of him ditching Taiwan is far greater.  There is money to be made selling EU sourced FPGAs.  

But sadly no one wants to hire me to do this either.   Back to Gaussian logarithms.

chili-chips-ba, 2025-02-22:
> "... EU sourced FPGA..."

is made in an UAE-owned fab, and also eying TSMC 😉 

---
> "... a fraction, about 1/4, of what is required for RISC-V on 4-LUTs..."

@tarik-ibrahimovic [tests](https://github.com/chili-chips-ba/openCologne/tree/main/8.StressTest/1.corescore_cc) have yielded substantially different results:
[image]
By all means, please go for the challenge to prove Tarik wrong. The more eyes, brains and perspectives on the problem, the better the final conclusion.

---
> "...  I could calculate it exactly..."

Please, don't calculate. Just do it (Nike style). Push it through the tools, then compare the numbers that the tools have calculated.

---
> "... ditching Ukraine, the risk of him ditching Taiwan is far greater..."

The USA seems to be ditching the EU as a whole, so that it can focus on the Far East (that Taiwan is a rather small part of).

PythonLinks, 2025-02-22:
I do not think that the core score is a good measure.  Serve is a bit serial processor.  It is not at all clear to me how that is implemented.   I suspect that it initially targeted  ICE40 LUT-4 devices.  reportedly it takes 32 or 64 clock cycles to execute one instruction, plenty of time to load registers from BRAMS.   What is needed is a regular RISC-V that tries to multipliplex 32 x 32 bit registers.    

Checking "[Awesome Gatemate](https://github.com/PythonLinks/awesome-gatemate)" There are three RISC-Vs on Gatemate.   When I started looking into these issues, on Dec 28th, 2024, one of the two FemtoRV authors said that "Without pipelines given the memory model, two clock cycles per instruction is possible when using quark-bicycle."   Looking closer, it has a full register set. 

>  reg [31:0] registerFile [31:0];

[Femto RV Bicycle](https://github.com/BrunoLevy/learn-fpga/blob/master/FemtoRV/RTL/PROCESSOR/femtorv32_quark_bicycle.v)

So that would be a good comparison.  Probably not the one that runs on GateMate now. 

The second RISC-V on GateMate is the NEORV32. From the [documentation](https://stnolting.github.io/neorv32/#_architecture)

> The data register file contains the general purpose architecture registers x0 to x31. For the rv32e ISA only the lower 16 registers are implemented.  

A web search says that. 

> "RV32E is a reduced version of RV32I for embedded systems, with 16 integer registers and soft-float calling convention. It uses the same instruction-set encoding as RV32I and can be combined with standard extensions."

Here are the notes on porting NEORV32 to gatemate.  

https://github.com/stnolting/neorv32/discussions/983

and

https://github.com/stnolting/neorv32-setups/tree/main/cologne_chip/GateMateA1-EVB

Sadly no resource utilizations are published.  If it is indeed the `i` version, and not the `e` version, that would be another good comparison. 

There is also the [EduBoss](https://fpga-ignite.github.io/presentations-pdf/presentation16.pdf) but sadly that page does not quote the resources for GateMate, and all the quoted examples store the register file in SSRAM.  But there is the option to store the register file in LUTRAM,  so that might make a good 3rd comparison.  Really all 3 comparisons should be done. 

Better yet do a 64 bit RISC-V.   

If we want to showcase the 4 bit multiplexers, we need to use examples that are actually based on using lots of multiplexers. 

> The USA seems to be ditching the EU as a whole, so that it can focus on the Far East (that Taiwan is a rather small part of).

Do we believe anything that the White House says?   I will remind you that MUSK has significant factories in China, and thus is subject to pressure from them.    IIRC Trump also has business interests in China.  For security, Taiwan needs to give him a nice beachfront Casino!  One that is not transferable!

chili-chips-ba, 2025-02-22:
... good point about the need to expand the portfolio of RISC-V cores used for GateMate Stress Testing👍 

Towards that, Tarik is trying to bring up [Veerwolf](https://github.com/chili-chips-ba/openCologne/tree/main/8.StressTest/4.veerwolves) on GateMate, thus far unsuccessfully 🤒 

@stnolting has also reported a number of problems with [NEORV32 on GateMate](https://github.com/stnolting/neorv32/discussions/983). Here is one related to the 8-input muxes.
[image]
There is a bunch of other issues. The entire text is a good read.

While Tarik shall add [femtorv32_quark_bicycle](https://github.com/BrunoLevy/learn-fpga/blob/master/FemtoRV/RTL/PROCESSOR/femtorv32_quark_bicycle.v) and [eduBOS5](https://fpga-ignite.github.io/presentations-pdf/presentation16.pdf) metrics to his GateMate test suite, it would also be very interesting to hear @BrunoLevy, @trabucayre, @matthias thoughts on this FPGA architecture benchmarking effort, including about where to take it in the next step.

PythonLinks, 2025-02-22:
> Tarik shall add [femtorv32_quark_bicycle](https://github.com/BrunoLevy/learn-fpga/blob/master/FemtoRV/RTL/PROCESSOR/femtorv32_quark_bicycle.v) and [eduBOS5](https://fpga-ignite.github.io/presentations-pdf/presentation16.pdf) metrics to his GateMate test suite.

Great.  Thanks for listening.  From a marketing perspective, what is needed is to also get the statistics for those soft cores on the $60 ECP5 MuseLabs IceSugar Pro, and the $38 Pico Ice with ICE40 UP5K.  

This data will be a big help for my talks about the advantages of stack machines vs RISC-V soft cores on FPGAs with LUT4s and no MUXs.

chili-chips-ba, 2025-02-22:
Board pricing is not a reliable indicator of the cost of FPGA device that a particular board carries. The FPGAs are generally categorized by their LUT count. The LUT count is therefore the primary entry point into all comparisons. 

We can also try to attach the Cost Factor, but only as the last/informal column, and with caveats. 

To that end, any recommendations from your side on the source of cost info? Should we use Digikey pricing for 1K quantities??

PythonLinks, 2025-02-22:
I agree that board pricing is not the best indicator. 

Digikey pricing for 1000 parts seems like a more reasonable metric.  

The larger point is not just port those soft cores to GateMate, but also to the price similar FPGAs.

Mecrisp, 2025-02-22:
> While Tarik shall add [femtorv32_quark_bicycle](https://github.com/BrunoLevy/learn-fpga/blob/master/FemtoRV/RTL/PROCESSOR/femtorv32_quark_bicycle.v) and [...]

The FemtoRV-Matthias here :-) Huh, the Gatemate FPGAs look nice, but these are quite large FPGAs. The FemtoRV32-Quark (and Quark-Bicycle) implementing RV32I only were designed to run in tiny Lattice HX1K FPGAs. I think for the quite large Gatemate FPGAs, FemtoRV32-Gracilis implementing RV32IMC with interrupts is a better match. I have, however, no idea on FPGA benchmarks.

Mecrisp, 2025-02-22:
By the way, here is a small RISC-V demo project featuring the FemtoRV32-Quark: https://codeberg.org/Mecrisp/Nandland-RISC-V The Quark-Bicycle is a drop-in replacement if you like.

Mecrisp, 2025-02-22:
Here is a larger port of the RISC-V playground for Lattice UP5K, featuring FemtoRV32-Gracilis:

https://badge.team/docs/badges/mch2022/software-development/risc-v/
https://github.com/badgeteam/mch2022-firmware-ice40/tree/master/projects/RISCV-Playground

Maybe also some of the other examples for the MCH2022 badge could be interesting.

chili-chips-ba, 2025-02-23:
... this discussion seems to be converging on a `configurable` [corescore ](https://github.com/chili-chips-ba/openCologne/tree/main/8.StressTest/1.corescore_cc) benchmark, one that would allow plug-and-play drop-in replacement of RISC-V cores within that multi-core SOC. 

That's to, in addition to the incumbent SERV, let the testing encompass additional cores, such as the ones that @Mecrisp and @PythonLinks have brought up.

PythonLinks, 2025-02-23:
I changed the title from Documenation to "Multiplexer Stress Test".  I wan to keep everyone's eyes firmly on the issue of the advantages of the GateMate multiplexers over LUT-4 Multiplexers.  

## Bicycle 

The Femto RV bicycle has that full multiplexer, and another large multiplexer for the 32 bit shifter, and "is quite small", meaning that it does not have much else.  So it is a perfect demo of the strength of the GateMate Multiplexers.   My only issue with bicycle is that it only has 1 shift multiplexer, not 2.   

>  Use the same shifter both for left and right shifts by applying bit reversal

https://github.com/BrunoLevy/learn-fpga/blob/master/FemtoRV/RTL/PROCESSOR/femtorv32_quark_bicycle.v#L128

## Serve 
As I suspected Serv is not a good stress test for multiplexers.  From the documentation, it is clear that it stores the register file in RAM. 

[image]

## VeerWolf 

VeerWolf is probably also not a good stress test/ marketing pitch.  It is a big cpu with lots of stuff besides register and shifter multiplexer.   

## Gracilis

I suspect that the same is true for the Gracillis one recommended by @Mecrisp.   Although it probably has two shifters.  My goal here is not to find the best RISC-V for GateMate.   My goal here is not to show what is wrong with GateMate, but to show what is right with it. To show that if you have an application with a high percentage of Multiplexers, what is the best low cost FPGA to use. 

And do not be shy about putting price front and center as the first column!   I hate to say it, but real world customers put price front and center.  They want that comparison.  Technical details, well they leave that to people like us who focus on technical details.   Still we need to keep in mind what those managers have in their minds.

BrunoLevy, 2025-02-23:
Hello, 
Thank you for the very interesting discussion !
If I understand well, the goal of the discussion is finding a reasonably simple risc-V core with a large number of muxes to demonstrate GateMate ? Here is a couple of ideas:
- Maybe we could start from the bicycle (that has a single shifter) and add a right shifter to it (but I'm unsure this would add a mux the way you want to demonstrate, since bicycle and the other FemtoRVs already have muxes to select from the direct and bit-reversed output of the shifter, but I have a very partial understanding on how this maps to muxes...).
- Another (more complicated) possibility is considering a pipelined core with register forwarding. This involves three-way muxes for selecting the input of the execute stage. An example is [here](https://github.com/BrunoLevy/TordBoyau) and associated [design notes](https://github.com/BrunoLevy/learn-fpga/blob/master/FemtoRV/TUTORIALS/FROM_BLINKER_TO_RISCV/PIPELINE.md). It is not as clean and self-contained as the Femto series though... (and lacks an I$ and D$ caches)
- One more point: if the goal is to create a tech demo, then I think it is interesting to have some interrupts and traps mechanism (like in FemtoRV-gracilis), since it makes it possible to have a very small core and emulate instruction sets in traps
- Another idea: starting from a Gracilis base, a minimal RV32F core (with a floating point register bank), implementation of MAC(multiply-add) in hw, and software traps for all other RV32F instructions. It would both add a number of muxes and create an interesting tech demo.

chili-chips-ba, 2025-02-23:
... while it is a bit funny to have to work this hard to find a design that CologneChip FPGA would excel in:
  1) knowing that GateMate is severely lacking on the LUTRAM front
  2) and looking to offset this shortcoming though muxing and math functions

a `minimal RV32F` core seems to be a good option. 
- How long would it take to put one together to benchmark GateMate against GowinSemi, Xilinx and Lattice?!

BrunoLevy, 2025-02-23:
I have a basic microprogrammed RV32F unit [here](https://github.com/BrunoLevy/learn-fpga/blob/master/FemtoRV/RTL/PROCESSOR/petitbateau.v), see also [design notes](https://github.com/BrunoLevy/learn-fpga/blob/master/FemtoRV/TUTORIALS/FPU.md). It would be mostly a matter of:
- removing FDIV and FSQRT from it (that eat-up of lot of resources and that could be emulated in a trap)
- grafting it on a gracilis
- make sure gracilis generates traps for the missing instructions
If @Mecrisp and I can find some time to work on this, I'd say it is a couple of weeks full time , but since we do that on our free time, I'd say something around 3 months. It also depends on how easy it is to talk to CologneChip FPGAs (if it speaks Yosys then it is easy).

Note: this is a quite unefficient multicycle FPU. It would be also possible to create a pipelined version, but this would require much more work / more time (I have been wanting to do this for more than 1y without finding the time...)

chili-chips-ba, 2025-02-23:
... the good news is that we can also count on the direct help from Cologne Chip engineering staff, represented on this discussion thread by @pu-cc

BrunoLevy, 2025-02-23:
I've taken a quick look at Cologne Chip whitepaper, it seems its CPE also has an interesting modular multiplier function block. If this one is fast, we could imagine creating an RV32D (double precision) multiply-adder.

BrunoLevy, 2025-02-23:
Other ideas for a technical demo:
if it is easy to create a large number of multipliers, having a GPU-like design would be interesting, either computing pixels in parallel and generating an image, or doing some AI/inference (and the low power consumption of the Cologne Chip could be an interesting argument).

chili-chips-ba, 2025-02-23:
We seem to have at this point assembled a stack of ideas for the benchmarking tangents that would _stack the deck_ for GateMate👍. It shall be both interesting and revealing to stack up:

__--1--__
   - GateMate math resources built into their unique LUT-Tree structure
   - to the [DSP hard macros](https://github.com/chili-chips-ba/openXC7-TetriSaraj/blob/main/99.more-boards/0.tang-nano-9k/0.doc/Gowin/UG287-1.3.3E_DSP.pdf) + generic LUTs found in the classic FPGA devices

__--2--__
   - GateMate 8-input logic/muxing capability
   - to the generic LUT with [MUXF7](https://docs.amd.com/r/2021.2-English/ug974-vivado-ultrascale-libraries/MUXF7) components routinely found in the mainstream architectures

In the process, we can also compare GateMate _power consumption_, _Fmax_ and _pricing structure_ to the others.

> @pu-cc, if you have any further guidance on this attack plan from the Cologne Chip insider perspective, this is the time to chime in.

BrunoLevy, 2025-02-25:
Dumped some notes [here](https://github.com/BrunoLevy/learn-fpga/blob/master/FemtoRV/TUTORIALS/Morph.md), @Mecrisp do not hesitate to tell us what you think !

PythonLinks, 2025-02-27:
I tried to fire up FemtoRV  Quark and then bicycle on Mac OS X, and ran into the following issue with Quark.  Basically does not work with the newest versions of Yosys.   There is a pull requrest which fixes the problem for the ICESTICK, but I expect that bicycle will not fit on ICESTICK.  It only has ~1260 LUTs. 

https://github.com/BrunoLevy/learn-fpga/issues/125

I am curious about the savings when using multiplexers over 4-input LUTs, but am not sufficiently motivated to debug the problems.   Really it is an issue for Cologne Chips to resolve, if they want to demonstrate the advantage of their FPGA over the competing Lattice devices. 

I will try it again in about a month.  Maybe the problems will be fixed by then.

chili-chips-ba, 2025-02-27:
> _"... it is an issue for Cologne Chips to resolve, if they want to demonstrate the advantage of their FPGA over the competing Lattice devices..."_

@PythonLinks, could you please open another ticket here for that issue, so that we can keep track of it.

tarik-ibrahimovic, 2025-03-11:
Tuning in with some numbers to this discussion. Benchmarking Gowin GW2AR-18C C8/I7 and CCGM1A1 with @aimamovic6's [Sigma-Delta DAC](https://github.com/aimamovic6/Sigma-Delta-DAC) yielded detailed insights which you can check out in depth [here]( https://github.com/chili-chips-ba/openCologne/tree/main/8.StressTest/7.sigma_delta_DAC#logic-utilization-in-sigma-delta-dac-implementations). 

### The takeaway  

Purely in the sense of logic capability, LUT-trees are on-par with LUTs when it comes to complex arithmetic (-28% of logic elements used for the same design), but a mentioned lack of DSP HMs really hurts the end results (+37% of logic elements used for the same design). However, when normalizing the number of logic elements by the number of configuration bits, it seems that total area on the chip is still larger in CCGM1A1 when compared to traditional LUT4 GW2AR-18C, with or without DSP HMs.

chili-chips-ba, 2025-03-14:
@PythonLinks, do you really believe that the 20K LUT [GW2AR-LV18QN88C8/I7](https://www.amazon.com/GW2AR-18-Development-64Mbits-Computer-Console/dp/B0C5XJV83K/ref=sr_1_2?crid=211FMET2UGITR&dib=eyJ2IjoiMSJ9.AEwgm-YPxdP261anDq1-BsiAoDJokec5dlu1rs_6y4N8VIMDQSVd_USSFpfV0BMYwL2vemaIXHF6cvAL-ZfoOsVl5J9WjIsC1jpSWGu_ozI4aP7jhdhIZ98mrOCBGseVkAemSZYCPfUt87aYMLWbXMUxEOzcgog_66SXvkaT7gkSv8RGEV7iH95c_aUNfic2TDO7NvLOAzdLOYmgKtUy_mCzTkgPy1Kg_zBYsnC_XYA.rAimm9RbvFQh8rquSimj9Fm43SKIbFy6guzKqM3nQG4&dib_tag=se&keywords=tang+nano+20k&qid=1741975442&sprefix=tang+nan%2Caps%2C326&sr=8-2) costs 69.10EUR?!

[image]

PythonLinks, 2025-03-14:
I was just quoting the mouser price. 
Thank you for the correction. 
Now if I could just find the data sheet to learn more about that Gowin FPGA.  
Christopher Lozinski

chili-chips-ba, 2025-03-14:
... Mouser is a hit and miss, esp. for the Far Eastern devices that they don't carry directly. 

You can get Gowin Arora doc pack from [here](https://github.com/chili-chips-ba/openCologne/tree/main/8.StressTest/0.doc/Gowin-Arora). One notable difference is that, while Arora does not have on-chip SerDes, it comes with in-the-package 8MB or 16MB RAM!

chili-chips-ba, 2025-03-14:
@PythonLinks, we are very interested to see your testcase, along with your results for the _GateMate-vs-Arora_ matchup!

The former is touted as a 20K/40K (8-input/4-input) FPGA, while the latter is the straight 20K LUT4 device. The former has a steeper price tag, but it comes with 5Gbps SerDes. The latter, on the other hand, includes IOSERDES and 8MB SDRAM that can host a large Frame Buffer for 1080P@60Hz HDMI. These two differentiating features (SerDes and SDRAM), while of interest when building a product, are not the subject of this benchmarking.

To provide the complete context, @nsokolovic7 is also porting @mirekez [PNR_TESTS](https://github.com/mirekez/pnr_tests) to both GateMate and Arora. We are eagerly awaiting his results too...

PythonLinks, 2025-03-15:
Thank you for the link to the documentation directory, and for the comparative analysis. 
I created a README for that directory, I have permission to accept my own pull request, but it is more professional if someone else reviews it. 

I took the time to read much of the documentation.  If I understood it correctly, Gowin has Configurable Function Units (CFU)and configurable Logic Units(CLU).  Both CFU and CLU have a routing unit, and four configurable logic sections (CLS).   Each CLS has 2 x 4-input LUTs.   The difference between the CFU and the CLU is that the CLSs in the CLUs cannot be configured as SRAMs.  

I believe that a CFU and CLU can operate as a 4 input MUX, which would consume 4 CLSs or 8 x 4-input LUTs.  Meaning that GateMate is more efficient at implementing RISC-V multiplexers.  Of course I could not find the detailed circuit diagrams, so I am not quite sure that i understood it all correctly.   

Barrel shifters also use multiplexers.  There is mention of a barrel shifter in the Gowin hard core DSP blocks, but the DSP documentation did not give much information.  It mentions Shift Input A (SIA), and Shift Input B (SIB), they are shown in the DSP circuit diagram, and in the table of ports.  

What is really needed is for someone to take the time to build the FemtoRV bicycle on both the Gowin board and a GateMate board, and compare the actual numbers. 

An interesting project.

chili-chips-ba, 2025-03-15:
... yes, the _proof is in the pudding_ 😉 Will you be able to built it?!

PythonLinks, 2025-03-15:
Well I have the technical skills to do this.   I am very good at doing whatever it takes to get things to work. It just takes time.  I have the time to do it.  Let us discuss this further privately.  I am @Christopher Lozinski on the Radiona Discord Server.  I sent a private message to @ChilliChips on Discord. 

If my preliminary analysis is correct, and it really takes a whole  CFU or CLU with 8 x 4-input LUTs to build a Gowin 4 input multiplexer, then GateMate would have a serious competitive advantage in RISC-V register multiplexing.

chili-chips-ba, 2025-03-15:
... 8080 with VGA for [SpaceInvaders](https://github.com/chili-chips-ba/openCologne/tree/main/8.StressTest/8.spaceinvaders-fpga) from the 80's:

[image]

tarik-ibrahimovic, 2025-03-16:
FemtoRV32-quark-bicycle ported, no peripherals, just RISC-V. Take a look at the [figures](https://github.com/chili-chips-ba/openCologne/tree/main/8.StressTest/10.femtorv32-quark-bicycle#resource-utilization), and if you still have doubts the sources are there, go ahead! 

In short:
-  GW2AR-18C : 1019 (877 LUT, 142 ALU) / 20736 (~5% usage) 
-  CCGM1A1: 1116 CPEs / 20480 (5.4% usage)

DFFs and BRAMs are the same.

PythonLinks, 2025-03-16:
I am impressed.  That was fast.  But it violates my instincts.  The regular FemtroRV uses about 1100 ICE40 4-input LUTs.  Bicycle should be using more than that.  Even on GateMate, 11 multiplexers / bit * 32 bits * 2 registers = 704 CPEs.  That is just for the multiplexer.   Plus the rest of the RISC-V.   If I read the docs right, the Gowin should be using a lot more.   And why are they both using 2 memory blocks when they do execute in place (XIP) ?  So I took [a closer look at the code](https://github.com/chili-chips-ba/openCologne/blob/main/8.StressTest/10.femtorv32-quark-bicycle/1.hw/femtorv32_quark_bicycle.v#L339). 

My apologies.  They are both storing registers in a register file in memory.  What we want is a RISC-V soft core that actually stores registers in registers, not in memory.  That will run faster.  It also explains why @Mecrisp said that the bicycle soft core takes 2 clock cycles per instruction.   @BrunoLevy 

That is also doing dual ported memory for the register file, but I thought you just said that the 8080 analysis said that Gowin "may not offer the same dual port capability."

Regards
Christopher Lozinski

tarik-ibrahimovic, 2025-03-16:
Gowin is using Semi-Dual Port Block RAM (SDPB), which is fully supported. To be fully specific, Gowin only doesn't support read-before-write mode in dual-port mode, which CologneChip does (which is required in the [spaceinvaders-fpga](https://gitlab.com/x653/spaceinvaders-fpga/-/blob/main/mem.v?ref_type=heads#L27) by default). This isn't required in this case since there are no concurrent reads and writes on the two ports.
[image]

As for the non-intuitiveness of these results, I invite you to take a look at this numerical [analysis](https://github.com/chili-chips-ba/openCologne/tree/main/8.StressTest/6.sw_analysis) of the GateMate LUT-trees. From there it's obvious that LUT-trees aren't a match for full LUTs in terms of logic. Also, have in mind that the tools play a big role in this.

PythonLinks, 2025-03-16:
Thank you for explaining the limitations of Gowim memory. 

The inefficiency of LUT Trees totally makes sense to me.  Since the registers are being stored in memory,  there are no big multiplexers, I would expect the Gowin implementation to be smaller.  My point was that this example violated my expectations for a multiplexer demo.  I then discovered that this example does not use two large multiplexers for the registers.   It just stores the registers in memory.   While LUT trees are a weakness of GateMate, the accompanying multiplexers are a strength.  

It would be really beneficial to Cologne Chips to demonstrate the advantages of the multiplexers with a RISC-V soft core that actually uses multiplexers.   And then it would be great to see a benchmark of the register based GateMate RISC-V soft core and similarly sized GOWIN memory based RISC-V soft core.

tarik-ibrahimovic, 2025-03-16:
I get the point that there's room for pushing the multiplexing usage, and we'll do our best to find out what GateMate can do. RISC-V cores feature wide muxes for writeback and PC control and some for input and output data, but that's where it ends pretty much. Register Files are implemented either in LUTRAM or Block RAM for every core that I have seen (occupying 1024 registers and wasting logic on decoding and multiplexing isn't the best way to go around).

PythonLinks, 2025-03-16:
You were correct that  the main stream RISC-V softcores use RAM and not registers. 
A very interesting observation. 

But the bicycle takes at least two clock cycles per instruction, half as fast as using registers.    Stacking the deck in terms of LUTs is not fair.  But stacking it in terms of performance/LUTs is perfectly fair.  Even good marketing. 

More importantly it is critical to find a market niche for the GateMate, or they may as well go home.

If you do a search of "RISC-V Verilog single cycle", there are a ton of choices.  The question is which one is good?  Maybe look at the github stars?   

Presumably VEX-RISC can also do this, but not out of the box, and scala is a tall scary stack to learn and debug.  Vex Risc is really for complex pipelined architectures, I am more interested in the smaller devices I suspect that the ones that use registers target six input luts, and  have a lot more that they do.  Not simple.    On the other hand it is a great brand name.  A good skill to learn.   And presumably a great idea for gatemate to support multiple VEXRISC versions. 

If you look at the [VexRisc page](https://github.com/SpinalHDL/VexRiscv)
The first one that does not fit on ICE40 looks interesting.  The preceeding ones fit on ICE40, the last one does not, suddenly its benchmark speeds go up.   If it has a single cycle barrel shifter then presumably all operations are single cycle.  

```
VexRiscv small and productive with I$ (RV32I, 0.70 DMIPS/MHz, 4KB-I$)  ->
    Artix 7     -> 220 MHz 730 LUT 570 FF 
    Cyclone V   -> 142 MHz 501 ALMs
    Cyclone IV  -> 150 MHz 1,139 LUT 536 FF 
    iCE40       -> 66 MHz 1680 LC

VexRiscv full no cache (RV32IM, 1.21 DMIPS/MHz 2.30 Coremark/MHz, single cycle barrel shifter, debug module, catch exceptions, static branch) ->
    Artix 7     -> 216 MHz 1418 LUT 949 FF 
    Cyclone V   -> 133 MHz 933 ALMs
    Cyclone IV  -> 143 MHz 2,076 LUT 972 FF 
```

chili-chips-ba, 2025-03-16:
> "... it is critical to find a market niche for the GateMate, or they may as well go home..."

@PythonLinks since you're the one who is repeatedly making GateMate Mux efficiency statements (granted, based on theoretical hypothesizing), you seem to be in the best position to create a testcase that would empirically proves it. 

Stack the deck as much as you want. A synthetic proof is still a proof and a good starting point for further discussion...

pu-cc, 2025-03-16:
> FemtoRV32-quark-bicycle ported, no peripherals, just RISC-V. Take a look at the [...]

Thank you for pushing the test case today, I really appreciate that and have just had a look at it.

In my opinion, we need to consider an important characteristic of CPEs and interpret the statistics correctly: Put simply, and as you all know, CPEs consist of two independently configurable halves, each with 4 inputs. This results in 40,960 CPE halves for configuration and use, or 20,480 CPEs with up to 8 inputs. The statistics in the P&R logs do **not** take into account whether only half a CPE or a full CPE is configured -- it counts half configured CPEs as fully configured CPEs. This means that the number in the report also includes unused halves.

I admit that this is not clearly expressed in the utilization report. I just went over the netlist, and >70% of the design consists of half-configured CPEs. We will make sure that the report is clearer, I'll check and get back to you on this tomorrow.

chili-chips-ba, 2025-03-16:
@pu-cc, it sounds like you plan to get away with generic CPE, and introduce CPE8 and CPE4 concept instead of it. That's a move in the right direction.

- Can you also think of a metric to add to the report for the utilization of routing resources? 

In essence, if a design contains a bunch of left-over CPE4s, it'd be nice to also provide an insight into how accessible, or not, they are...

tarik-ibrahimovic, 2025-03-17:
@pu-cc that sounds promising. How do you comment the [fpga_torture](https://github.com/chili-chips-ba/openCologne/tree/main/8.StressTest/5.fpga_torture) achieving only 28655 instantiations of LUT3(xor) + DFF. This is a best case for routing and a single L2T4 should be able to host a 3 input XOR, meaning the number should approach the available 40960 L2T4s + DFFs?

Since that's the case where I've seen the most utilization, should we say that 30% of the chip is never utilized? DFF usage in that test is exactly 70%, and adjusting for this CPE4 abstraction, CPE usage would drop from the initial 96% to just 70% (that is in line with the number of DFFs so it would make sense)

As @chili-chips-ba said key is in the "how accessible" are the CPE4s.

chili-chips-ba, 2025-03-19:
... @eniokaljic observed that the CPUs, esp. the microcontrollers, are mostly muxes, with a bit of decoders and math functions. As such, they seem good testbeds for bringing the best out of GateMate muxing and math LUT-tree structures.

PythonLinks, 2025-03-19:
I am glad to hear that I am not the only one who thinks that.  
I would love to read the source for that reference.

I took a closer look at VexRISC-V.

> An FPGA friendly 32 bit RISC-V CPU implementation.

They have 39 different pre-configured versions. 

**RegFilePlugin**

 > This plugin implements the register file.

**Parameter regFileReadyKind**

> Can be set to ASYNC or SYNC. Specifies the kind of memory read used to implement the register file. ASYNC means zero cycle latency memory read, while SYNC means one cycle latency memory read which can be mapped into standard FPGA memory blocks.

So I did a search on the VexRisc github repository on "regFileReadyKind plugin.async".  Only 1 example soft core out of 38 examples used the async option. And that was the multicore example.   Meaning they are all waiting 1 extra cycle to read a register from memory, two extra cycles if they are on single port RAM.   

WOW.  So basically all of the VexRISC-V soft cores toss out half of the performance, because multiplexers are too expensive!

pu-cc, 2025-03-20:
> [@pu-cc](https://github.com/pu-cc) that sounds promising. How do you comment the [fpga_torture](https://github.com/chili-chips-ba/openCologne/tree/main/8.StressTest/5.fpga_torture) achieving only 28655 instantiations of LUT3(xor) + DFF. This is a best case for routing and logic, a single L2T4 should be able to host a 3 input XOR, meaning the number should approach the available 40960 L2T4s + DFFs?
> 
> Since that's the case where I've seen the most utilization, should we say that 30% of the chip is never utilized? DFF usage in that test is exactly 70%, and adjusting for this CPE4 abstraction, CPE usage would drop from the initial 96% to just 70% (that is in line with the number of DFFs so it would make sense)
> 
> As [@chili-chips-ba](https://github.com/chili-chips-ba) said key is in the "how accessible" are the CPE4s.

I uploaded an initial draft that prints more information about "CPE4" and "CPE8" usage. You also now see more information on the consumption of routing resources, with `SB_BIG_MUX` and `SB_SML_MUX` being the relevant information for you.

https://colognechip.com/downloads/testing/cc-toolchain-win.zip
https://colognechip.com/downloads/testing/cc-toolchain-linux.tar.gz

Regarding [fpga_torture](https://github.com/chili-chips-ba/openCologne/tree/main/8.StressTest/5.fpga_torture), you will see in the report that P&R sometimes use CPEs for routing as well, we call them "bridges", using the built-in MUX8. It is not possible to make a general statement about the number of these elements in a design.

chili-chips-ba, 2025-03-20:
> "... P&R sometimes use CPEs for routing as well, we call them "bridges", using the built-in MUX8..."

This reminds us of Efinix [Quantum Fabric](https://www.efinixinc.com/technology.html), which is about _eXchangeable Logic and Routing_ (XLR) concept. 
- @pu-cc, where can we get more info about it for GateMate?!

pu-cc, 2025-03-20:
> @pu-cc, where can we get more info about it for GateMate?!

There is actually not too much to tell about it. It is possible to use the MUX8 built into the CPE to route signals from one routing plane to another, if the routing structure is already occupied at a certain position. More precisely, a signal can be routed from one of the 8 inputs to one of the outputs by permanently assigning the select inputs of the MUX. In terms of timing, it is of course less performant, which is why one should actually try not to implement bridges in the critical paths during P&R.

chili-chips-ba, 2026-10-21:
Thanks for the benchmarks and the discussion. These numbers come from the legacy p_r flow, which we no longer use: openCologne now builds with the open Peppercorn flow (Yosys + nextpnr-himbaechel + gmpack). Closing; please open a new issue with Peppercorn results if the question still stands.

closed 2026-10-21 by chili-chips-ba