chili-chips-ba/openCologne #59: Clarification on timing p_r option results

tarik-ibrahimovic, 2025-03-01:
For a variety of cases ([corescore](https://github.com/chili-chips-ba/openCologne/tree/main/8.StressTest), [fpga_torture](https://github.com/chili-chips-ba/openCologne/tree/main/8.StressTest/5.fpga_torture), and [LUTRAM_test](https://github.com/chili-chips-ba/openCologne/tree/main/8.StressTest/2.LUTRAM_stress_test)) there was a trend of best timing setting in p_r  `-tm 1` also achieving **best routability!** This is counterintuitive since usually there's a tradeoff between timing and routability. I am hereby seeking if there is any intuition behind these results and is this predictable in any way. @pu-cc

chili-chips-ba, 2025-03-01:
... for context, here is the relevant excerpt from GateMate [datasheet](https://colognechip.com/docs/ds1001-gatemate1-datasheet-latest.pdf) (Feb.2025, page 131).

[image]

pu-cc, 2025-03-05:
The same as mentioned in https://github.com/chili-chips-ba/openCologne/issues/62 applies here.

By default, timing reports are generated for the worst corner/lowest temperature. With better conditions, timings also get better; Under certain circumstances, timing violations can also occur, or do no longer occur, which could result in different netlists and routability. This is to be expected and is not counter-intuitive.

chili-chips-ba, 2025-03-05:
> _"... worst corner/lowest temperature..."_

Is there an insider wisdom we are missing in the above tying of the worst corner to the lowest temperature?!

While the effect of `temperature inversion` is present in smaller process geometries, is this to say that GateMate is subject to it too?

closed 2025-07-12 by tarik-ibrahimovic