chili-chips-ba/openCologne #62: Unexpected Fmax sensitivity in fpga_torture

tarik-ibrahimovic, 2025-03-01:
Since fpga_torture design is mostly linear by nature (see the image below), it's unclear why for different different `-tm` values `p_r` yields such different results in terms of Fmax. It's expected that the tools always converge on a similar solution here, for this relatively simple design. Is there any wisdom behind these results?

>Results for `-tm={3,2,1}`
>141.12 MHz at -40°C (which is not the case)
>200.04 MHz at 25°C and
>262.67 MHz at 125°C

[image]

pu-cc, 2025-03-05:
It seems that the function behind the various timing settings for P&R has not been fully understood.

Process variations and temperature have an influence on the performance of the silicon.
In your case, the bitstream of the design will probably not change. It is exactly the same configuration, except that different corners and temperatures were taken into account when creating the report. So it is completely to be expected.

tarik-ibrahimovic, 2025-03-05:
If I understand correctly, the -tm timing option is used to specify the operating environment temperature as a target for the Static Timing Analysis (STA) to consider. However, the routing algorithm itself operates independently of -tm and does not directly account for this parameter. Is that correct? 

If so, are the better results in terms of routability over the reported examples obtained just by chances and reiterating the algorithm until STA is satisfied? And what about the chip not working in `-tm 1` highlighted in [issue61](https://github.com/chili-chips-ba/openCologne/issues/61)?

chili-chips-ba, 2025-03-05:
> _"... process variations and temperature ... different corners and temperatures were taken into account when creating the report..."_

As @tarik-ibrahimovic observed, the above indeed sounds like the `-tm` option was used primarily for STA. That brings another question: Given that GateMate comes in only one speed grade, i.e. CologneChip does not sort the silicon dies based on process variation, why is the `-tm` option needed at all? The tool should in such a situation iterate internally and cover both WC and BC. 

> _"... it seems that the function behind the various timing settings for P&R has not been fully understood..."_

The initial impression is that the `-tm` option is to guide the PAR to optimize for timing, and not to set the silicon operating condition. If there is a write up that explains it better, beyond the datasheet excerpt below, let's add it to this discussion thread.

[image]

chili-chips-ba, 2026-10-21:
Legacy p_r -tm option; pu-cc explained it sets the STA corner, not the implementation. Not applicable to the open Peppercorn flow we now use.

closed 2026-10-21 by chili-chips-ba