chili-chips-ba/openCologne #63: Is there an option for modular (as opposed to flat) implementation flow with Gate?!

tarik-ibrahimovic, 2025-03-01:
Corescore, as fpga_torture, again showed particular sensitivity in terms of Fmax when changing the p_r `-tm` parameter. Again, as fpga_torture, this is a copy-and-paste design: a number of self-contained SoCs with a large output mux. 

Here are the results:
- `-tm 1` -- 56 cores instantiated with a 41 MHz Fmax -- functioning at 16 MHz -- **NO** 
- `-tm3` -- 54 cores instantiated with a 21 MHz Fmax -- functioning at 16 MHz -- **YES** 

This is again highly unexpected since the design is modular and self-contained. Is there an option to "harden" the design and do the pnr accordingly?

chili-chips-ba, 2025-03-01:
... a 2-step flow would suffice too, where Step#1 could resolve the building block within itself, providing relative placement and routing, and Step#2 would only connect together multiple instances of that already "semi-hardened" block. 

Such **modular flow** would also open the path to creating pseudo-HMs, looking to bridge the gap left by the absence of the ready-made components such as IOSERDES and DSP macros...

chili-chips-ba, 2026-10-21:
Tracked upstream for the open flow: https://github.com/YosysHQ/prjpeppercorn/issues/29