chili-chips-ba/openCologne #65: Pre-built yosys doesn't support tcl scripts

tarik-ibrahimovic, 2025-03-05:
Pre-built `cc-toolchain-linux` yosys doesn't support tcl scripts, which makes it incompatible with FuseSoC! 

From the [documentation](https://yosyshq.readthedocs.io/projects/yosys/en/stable/cmd/tcl.html):
```
yosys> help tcl

tcl <filename> [args]

This command executes the tcl commands in the specified file.
Use 'yosys cmd' to run the yosys command 'cmd' from tcl.
```

And with the latest CC pre-built toolchain:
```
yosys> help tcl
No such command or cell type: tcl

```

pu-cc, 2025-03-05:
As far as we remember, there were problems with cross-compiling the `tcl` extension for Windows. But I can certainly take a look at the current status.

So currently, it is not part in the pre-built yosys for Windows and linux that we provide, but there are other solutions if you want to use it:
* you can build it yourself
* did you try the official pre-build yosys binaries, or the one offered by your package manager?

tarik-ibrahimovic, 2025-03-05:
I usually use a source-built Yosys, but while setting up a new machine recently, I wanted to speed up the process with pre-built binaries. This is more of a feature request for compatibility with FuseSoC, which integrates the gatemate toolchain out of the box. However, to use it, you currently have to build Yosys from source, which can be inconvenient for first-time users.

closed 2025-07-12 by tarik-ibrahimovic