chili-chips-ba/openCologne #66: Yosys LUT primitives translation into CC primitives -- potentially underperforming?

tarik-ibrahimovic, 2025-03-06:
Here are the results of the [veerwolfs](https://github.com/chili-chips-ba/openCologne/tree/main/8.StressTest/4.veerwolves) test. First of all, list of `CPE Component Statistics` seems to contain only simple logic functions, supposedly de-composable into L2T4 primitives (OR, AND etc...). Given that `Sum of COMB=37715` and the number of L2T4 primitives instantiable is `40960` in CCGM1A1, there's a premise that CCGM1A1 can fit this design. However, logic mapping to CPEs results in 20% CPEs more than avaliable. 

Yosys predicts 14k LUT4s, along with 8k LUT3 and 4k LUT2, summing to be 26568 LUTs (not even LUT4s). Considering the worst case: that all of these would be LUT4, 40960 L2T4s should cover them? 

How close to optimal would you say LUT mapping to CC primitives is? Is there anything I'm missing in this analysis? 
> yosys.log
```
   Number of cells:              41595
     $print                          1
     $scopeinfo                   2229
     CC_ADDF                      3055
     CC_BRAM_40K                    14
     CC_BUFG                         1
     CC_DFF                       8910
     CC_IBUF                        19
     CC_LUT1                       184
     CC_LUT2                      4269
     CC_LUT3                      7894
     CC_LUT4                     14221
     CC_MULT                         3
     CC_MX4                        777
     CC_OBUF                        17
     CC_PLL                          1
```
> pnr.log
```
CPE Component Statistics:
          OR      3202         8%
         AND     17296        45%
         XOR      1553         4%
       ANDOR         1         0%
       ORAND      5170        13%
       ICOMP       260         0%
        MX2a      3873        10%
        MX4a      2764         7%
      ANDXOR       146         0%
        ADDF       182         0%
       ADDF2      2458         6%
       ADDFx         7         0%
      ADDF2x        42         0%
        MULT       684         1%
      MULTFa        13         0%
      MULTFb        11         0%
      Route1        53         0%
              --------
Sum of COMB:     37715

           D      8879        95%
         DST        32         0%
       C_0_1       268         2%
      RAM_I1        70         0%
      RAM_I2        59         0%
              --------
Sum of SEQ:       9308

Sum of all:      47023

Number of CPEs on layer    1 / 2:   8828 / 8444
Number of CPEs using layer 1 + 2:  16356
Number of needed CPEs        / available CPEs:     25184 / 20480
Number of needed IO-Pins     / available IO-Pins:     36 / 144
ChipSize:     X: 160  Y: 128

FPGA too small for circuit!
```

chili-chips-ba, 2026-10-21:
Re-run on PrjPeppercorn (Yosys 0.69 `synth_gatemate -luttree -nomx8`, nextpnr-himbaechel 0.11.1-77): one VeeRwolf core needs 21852 / 20480 CPEs (107%; nextpnr reports 43703 / 40960 half-CPEs). Legacy `p_r` needed 25184 (123%). Better, but it still does not fit. Open question stays: why ~26.6k Yosys LUTs need more than 20480 CPEs. Details: https://github.com/chili-chips-ba/openCologne/tree/main/8.StressTest#prjpeppercorn-re-runs

chili-chips-ba, 2026-10-21:
On paper one core fits at 87.5% (35840 / 40960 half-CPEs). nextpnr packing overhead (3556 FF-only halves, 2410 from unpaired adders, and more) takes it to 107%. Table: https://github.com/chili-chips-ba/openCologne/tree/main/8.StressTest#veerwolf-1-core-66

**Upstream:** https://github.com/YosysHQ/prjpeppercorn/issues/30