chili-chips-ba/openCologne #73: Nextpnr with a fixed seed unpredictable Fmax behavior

tarik-ibrahimovic, 2025-07-12:
I've raised a more detailed [issue](https://github.com/YosysHQ/prjpeppercorn/issues/4) in project peppercorn repository. Nextpnr behaves chaotically even for a fixed seed, and for some frequencies gets stuck in infinite operation.

chili-chips-ba, 2025-07-12:
@mmicko any thoughts?

Ravenslofty, 2025-07-15:
```
Traceback (most recent call last):
  File "/home/lofty/openCologne/9.BetrustedSOC/betrusted_soc.py", line 245, in <module>
    ret = main()
  File "/home/lofty/openCologne/9.BetrustedSOC/betrusted_soc.py", line 199, in main
    soc = BetrustedSoC(
        platform,
    ...<3 lines>...
        app_uart=args.app_uart,
    )
  File "/home/lofty/openCologne/9.BetrustedSOC/betrusted_soc.py", line 112, in __init__
    self.submodules.crg = CRG(platform, sys_clk_freq)
                          ~~~^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/lofty/openCologne/9.BetrustedSOC/betrusted_soc.py", line 86, in __init__
    self.clock_domains.cd_sys = cd_sys = ClockDomain()
                                         ~~~~~~~~~~~^^
  File "/home/lofty/openCologne/9.BetrustedSOC/deps/migen/migen/fhdl/structure.py", line 710, in __init__
    raise ValueError("Cannot extract clock domain name from code, need to specify.")
ValueError: Cannot extract clock domain name from code, need to specify.
```

I can't get it to build. can somebody tarball the output verilog etc for me to use?

tarik-ibrahimovic, 2025-07-19:
I am travelling, I'll get back to you tomorrow.

tarik-ibrahimovic, 2025-07-21:
@Ravenslofty Just to verify the build process, I just tried to build a clean pull in a python environment as the readme suggests and it worked for me, did you follow the same procedure?

Here is the tarball.

[archive.tar.gz](https://github.com/user-attachments/files/21345913/archive.tar.gz)

chili-chips-ba, 2026-10-21:
Re-run on current nextpnr-himbaechel (0.11.1-77, seed 1), BetrustedSoC re-synthesized with Yosys 0.69, PLL target varied:

| PLL target | Fmax | Run time |
|---|---|---|
| 10 MHz | 19.75 MHz (PASS) | 10.6 min |
| 28 MHz | 19.75 MHz | 10.8 min |
| 30 MHz, 2 runs | 19.75 MHz | 10.7 min each |
| 35 MHz | 19.75 MHz | 10.7 min |

No hangs. Same seed gives byte-identical output. Placement is the same for every target: with one clock, nextpnr steers by relative delay, so that is expected. This SoC tops out at about 20 MHz on GateMate. Also closed upstream as YosysHQ/prjpeppercorn#4. Details: https://github.com/chili-chips-ba/openCologne/tree/main/8.StressTest#prjpeppercorn-re-runs